forkineye/ESPixelStick #484: Web GUI, Does not show all files on SD Card

m1anh, 2022-03-08:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
FW Version4.0-beta4 Build Date Nov 13 2021 - 14:53:28

**Hardware Version**
ESP8266  - Wemos D1 Mini

**Binary release or compiled yourself?**
Binary release

**Operating System (and version)**
Windows 10 Pro 10.0.19042 Build 19042

**Web Browser (and version)**
Chrome Version 98.0.4758.102 (Official Build) (64-bit)

**Access Point**
Draytek 2830n

**Describe the bug**
Web GUI only displays 1 or 2 of the many FSEQ files upload to the SD Card.
Files can be load onto the card using FPP Connect or Directly by inserting SD Card in PC and using File Explorer makes no difference.
Memory card is a OLE 32GB, H2Testw'ed, 90mb read 20 mb write. fat32
Otherwise, card is read fine and will play all the fseq files.

m1anh, 2022-03-12:
After some further investigation, it seems the problem is being caused by the inclusion of brackets () in one or more of the file names. FPP Connect will upload a file name with Brackets in it. After a filename containing bracket is present in the file structure, the ESP has some unpredictable behaviour. Only one, some or none of the file names are then displayed. File names in "Device Setup" - "FSEQ File To Play" are also effected. I was using the name "Wizzard - (Off).xsq". Removing the SD card, placing it in a PC and deleting the () file, seems to fix the issue. Although, () are legal file name characters and should ideally be allowed, or an error thrown.

MartinMueller2003, 2022-03-12:
Brackets "[]" or parenthesis "()"? Your text says brackets and your example uses parenthesis. I assume parenthesis. Special characters are a bit of a pita in a lot of systems. This at least gives me something to work with.

FYI: A single quote  '  is mishandled by the FPP (turns into a space) and handled properly on the ESP.

MartinMueller2003, 2022-03-12:
So I just checked my file list and I have files with both parenthesis and square brackets and there is no issue displaying the files. You can see the files in my sample file list output above and in the screen capture below. I think you need to send me the debug output for the failed list so I can take a closer look at it.

[image]

MartinMueller2003, 2022-03-12:
Are you by any change using Curly Braces {}? Those I have not tried.

MartinMueller2003, 2022-03-12:
Curly Braces work just fine as well

[image]

m1anh, 2022-03-12:
Hi. The term "parenthesis" is seldom used in the UK. they are called brackets(), Square Brackets[] and Curly Brackets{}. Sorry, I did not realize US calls them by a different name, you learn something everyday. For clarity was referring to ASCII 0x28  and 0x29. But this is a red herring. The issue is much simpler and makes a lot more sense. The file name actually causing the issue was:-"Wizzard - I Wish It Could Be Christmas Everyday (Official Music).fseq", but it's not the Brackets(), it's just the overall file name length, anything more than 63 Char in length and it fails to display and sometimes prevents other (Shorter name) files from being displayed. This seems to be dependent on the order the files are loaded onto the SD Card. Upload the oversized file 3rd, only the first two files are displayed. If the file name length is a FS or memory preservation requirement and 63 Char is plenty, it would be good if a warning was displayed (SD Contains files over 63 Char!) or truncate the names in the old Dos style ~001 when being displayed. But if possible, prevent the oversize filename(s) from blocking the display of the remaining files on the SD Card. Sorry the misleading information. I hope this helps.

MartinMueller2003, 2022-03-13:
It does but there is nothing I can do about the limit and I cant enforce any sort of trimming once the file has been written to flash. If you write using a computer, then the file will be corrupted before any function I can use to read the file name from the SD card. I could add some protection to prevent you uploading such a file to the ESP but that does not help with files written by a PC. Better to just acknowledge that the file names need to be short. The next problem would be when the FPP or xLights asks for the file to be played using that long name. I would have no way to adjust that name to a rhort form name when my code did not write the file in the first place.

m1anh, 2022-03-13:
I understand the issues, I would propose just putting a note / warning in the GUI that file names must not excess 63 Char.
Also, we could ask the xlights team if they would be prepared to place a length check when ESPixelstick is selected as a controller and files are uploaded by FPP Connect(Nice to have). But it's easy to avoid the issues once you know about it. Many thank for the fantastic support and great software, keep up the great work.

MartinMueller2003, 2022-05-03:
Comment is added to the file management page.

closed 2022-05-03 by MartinMueller2003