forkineye/ESPixelStick #488: ESPFlashTool- Question

thespica93, 2022-03-17:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
3.2

**Hardware Version**

**Binary release or compiled yourself?**
Binary

**Operating System (and version)**
Windows 10

**Web Browser (and version)**
Google Chrome

**Access Point**

**Describe the bug**
I tried to fill this out to the best of my knowledge. I am new to this. 
I purchased these Wemos boards from Amazon. Here is the attached link for reference. 
https://www.amazon.com/dp/B081PX9YFV?psc=1&ref=ppx_yo2_dt_b_product_details

I plugged a micro USB into them and connect to my PC. I see an indicator light flash briefly on them. I open up the ESPFlashTool  Software. I enter my SSID and Passphrase. However I don't see any options in the drop down for Com Port (Picture attached.) I shorted out D3 to Ground on the Board and hit reset to get it into firmware flash mode. 

Am I doing anything incorrectly? Is it possible a PC USB port is not providing adaquete power to fully power the board?

Thank you 
-Nick
[image]

forkineye, 2022-03-17:
Either you don't have the CH340 driver installed or you're using a charging cable which doesn't have the data lines connected.

thespica93, 2022-03-17:
Hey thanks for the response. Where can I find the CH340 driver? I didn't install that. So that may be my problem.

Thanks
-Nick

forkineye, 2022-03-17:
> Hey thanks for the response. Where can I find the CH340 driver? I didn't install that. So that may be my problem.
> 
> Thanks -Nick

Everything you need is here - https://forkineye.com/espixelstick-v3/