forkineye/ESPixelStick #493: Lolin ESP32 PRO No Longer Works

marksat666, 2022-03-20:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
FW Version4.0-ci2009861803

**Hardware Version**

Lolin ESP32 Pro
**Binary release or compiled yourself?**
release installed via Flash Tool

**Operating System (and version)**
Windows 10  Home
10.0.19044 Build 19044

**Web Browser (and version)**
MS Edge Version 99.0.1150.46 (Official build) (64-bit)

**Access Point**
Netgear R7800

**Describe the bug**
I can no longer use the SD card on my Lolin ESP32 PRO
("Lolin..." is no longer a Hardware option in the Flash Tool)
When I select:

- D32 PRO: constantly reboots
- ESP32 TTGO T8: constantly reboots
- Generic ESP32: works but does not see SD card
- ESP32 CAM: works but does not see SD card
- D1 Mini32: works but does not see SD card

Dropping back to an older version (that has the Lolin hardware option) works, so the hardware is OK.

Here is the console output for the D32 Pro:

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12776
load:0x40080400,len:3032
entry 0x400805e4

[    ESP] ESPixelStick v4.0-ci2009861803 (Mar 19 2022 - 21:33:15)
[    ESP] v4.4-beta1-308-gf3e0c8bc41
[FileMgr] Flash file system initialized. Used = 200704 out of 327680
[FileMgr] SD Card Size: 15238MB
[FileMgr] > System Volume Information
[FileMgr]   IndexerVolumeGuid - 76
[FileMgr] Configuration File: '/config.json' loaded.
[    ESP] Processing Flash Tool config
[NetworkMgr] Using old style WiFi Settings
[  1653][E][vfs_api.cpp:102] open(): /littlefs/output_config.json does not exist, no permits for creation
E (1679) rmt: rmt_tx_stop(274): RMT CHANNEL ERR
E (1680) rmt: rmt_tx_stop(274): RMT CHANNEL ERR
E (1681) rmt: rmt_tx_stop(274): RMT CHANNEL ERR
E (1682) rmt: rmt_tx_stop(274): RMT CHANNEL ERR
E (1807) rmt: rmt_tx_stop(274): RMT CHANNEL ERR
E (1809) rmt: rmt_tx_stop(274): RMT CHANNEL ERR
E (1810) rmt: rmt_tx_stop(274): RMT CHANNEL ERR
E (1812) rmt: rmt_tx_stop(274): RMT CHANNEL ERR
E (1836) rmt: rmt_tx_stop(274): RMT CHANNEL ERR
E (1837) rmt: rmt_tx_stop(274): RMT CHANNEL ERR
E (1837) rmt: rmt_tx_stop(274): RMT CHANNEL ERR
E (1838) rmt: rmt_tx_stop(274): RMT CHANNEL ERR
E (1878) rmt: rmt_tx_stop(274): RMT CHANNEL ERR
E (1879) rmt: rmt_tx_stop(274): RMT CHANNEL ERR
E (1879) rmt: rmt_tx_stop(274): RMT CHANNEL ERR
E (1880) rmt: rmt_tx_stop(274): RMT CHANNEL ERR
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4009504c  PS      : 0x00060d33  A0      : 0x80095861  A1      : 0x3ffbc220  
A2      : 0x3f800014  A3      : 0xffffff5d  A4      : 0x3ffca900  A5      : 0x00060d23  
A6      : 0x00060d20  A7      : 0x00000001  A8      : 0x0000001f  A9      : 0x0000000f  
A10     : 0x3f800810  A11     : 0x00000014  A12     : 0x00000000  A13     : 0x00060d23  
A14     : 0x00060d23  A15     : 0xfffffffc  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xffffff61  LBEG    : 0x40087021  LEND    : 0x40087029  LCOUNT  : 0x00000027  

Backtrace:0x40095049:0x3ffbc2200x4009585e:0x3ffbc240 0x40095ab0:0x3ffbc260 0x4008541e:0x3ffbc280 0x4008549a:0x3ffbc2b0 0x40085135:0x3ffbc300 0x40167da3:0x3ffbc320 0x401706ef:0x3ffbc340 0x4016dc09:0x3ffbc360 0x4013e2d3:0x3ffbc380 

ELF file SHA256: 0000000000000000

Rebooting...

MartinMueller2003, 2022-03-21:
FYI: The log is showing that there is no problem accessing the SD card. The error seems to be in creating the initial config file.

Please use the attached Flashtool to upload the image to the ESP. Once you try to load the image, click save logs and copy the log file into this chat.
[ESPixelStickV4Dist.zip](https://github.com/forkineye/ESPixelStick/files/8312603/ESPixelStickV4Dist.zip)
.

marksat666, 2022-03-21:
Attached is the log file from the dev tool image load (using the D32 PRO hardware selection)..
[20220321_Log.txt](https://github.com/forkineye/ESPixelStick/files/8318224/20220321_Log.txt)
.

MartinMueller2003, 2022-03-21:
Thanks for the log. While there is still an issue, if you had filled in the SSID and password for your AP, the system would have come up. Not sure why you are getting the crash the first time the WiFi tried to come up, but the system looks stable after that.

MartinMueller2003, 2022-03-21:
This is not an issue in the latest builds.

marksat666, 2022-03-21:
Martin, I DID put in my WIFI info.  I tried it again with and without DHCP, but the logs look the same (it does save the wifi settings but I am not sure why it does not use them, config.json attached)..
[config.json.txt](https://github.com/forkineye/ESPixelStick/files/8319439/config.json.txt)
.

marksat666, 2022-03-22:
I got out a second Lolin to see if that first one had an issue, but the same thing is happening; when I flash the latest CI it just keeps rebooting after this message "Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled."

MartinMueller2003, 2022-03-23:
This is a sneaky little issue. Every time I close in on it, it changes. I have fixed a few minor issues and suspect there were a few timing related issues looking like a single problem. Still digging.

MartinMueller2003, 2022-03-24:
if you turn off relay support it should be stable. still tracking down what is wrong

marksat666, 2022-03-24:
Martin, I am not sure how to turn off relay support (I don't see any option like that in the Flash Tool)

MartinMueller2003, 2022-03-25:
I am working on a build that should get you moving. It will have relay turned off.

marksat666, 2022-03-26:
Martin, that latest version is working for me now (still having issues uploading sequences, but that is a prior issue).  I think you can close this one (or can I close it?).

MartinMueller2003, 2022-03-26:
The issue is NOT resolved. I simply turned off the I1C Relay support. And that still needs to be fixed.

MartinMueller2003, 2022-03-27:
I tracked one error down to an issue in the newer ArduinoJson implementation. There is still another issue I am tracking. This one is a real PITA

MartinMueller2003, 2022-05-01:
Took a break from this for a bit. I should get back to it next week.

MartinMueller2003, 2022-05-03:
Testing with latest code SD access is working fine.

closed 2022-05-03 by MartinMueller2003