forkineye/ESPixelStick #494: ESP32 (D1 format) SDCard files listed but mounting error

JavierCollazo, 2022-03-23:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
4.0

**Hardware Version**
3.0

**Binary release or compiled yourself?**
Yes

**Operating System (and version)**
Windows 10

**Web Browser (and version)**
Chrome

**Access Point**

**Describe the bug**
After powering the circuit, the SD card is read and files are listed, but afterwards a SDCard error arises and the web interface does not show the SD card options:

[    ESP] ESPixelStick v4.x-dev (Mar 23 2022 - 13:18:58)
[    ESP] v4.4-dev-2313-gc69f0ec32
**[FileMgr] Flash file system initialized.
[FileMgr] SD Card Size: 15223MB
[FileMgr] > System Volume Information
[FileMgr]   WPSettings.dat - 12
[FileMgr]   IndexerVolumeGuid - 76
[FileMgr] Arbol Espiral.fseq - 360072
[FileMgr] data1.txt - 77
[FileMgr] Configuration File: '/config.json' loaded.
[   536][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[   537][E][sd_diskio.cpp:790] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[   843][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00**
[FileMgr] No SD card installed
[FileMgr] Configuration File: '/output_config.json' loaded.
[FileMgr] Configuration File: '/input_config.json' loaded.
[WiFiMgr] WiFi Entering State: Connecting Using Config Credentials

MartinMueller2003, 2022-03-25:
That makes little sense. Let me get this straight:

- Compile with CS = 4
- Then configure CS to 5
- Card works fine

- Compile with CS set to 5
- SD Card does not work?
-

JavierCollazo, 2022-03-25:
Hi,

Thank you for your answer.

Yes that is what os happening. The funy thing is that after boot, card is
read and files listed, but afterwards the log says card error and the card
options do not show in the portal.

Maybe there are differences between the latest bin release and the current
source code that cause this issue.

I can try to compile the current source with no changes and config CS=5 to
see what happens.

Thanks again.

Best Regards,

Javier.

MartinMueller2003, 2022-05-03:
I am not seeing this and I ran a 24 play from SD card test that did not have any issues accessing the SD card.

MartinMueller2003, 2022-05-21:
Have you tried the latest build? We recently upgraded to the latest ESP IDF from Espressif. There were lots of changes. I have yet to experience the issue you are seeing.

MartinMueller2003, 2022-09-29:
There is an issue in the Flashtool that overwrites the SD card pin settings. That is why I do not see this issue. I use VSCode to upload my images.

MartinMueller2003, 2023-04-13:
This has been resolved in the flashtool

closed 2023-04-13 by MartinMueller2003