forkineye/ESPixelStick #495: ESP32 no SD card installer

gabodiazjayme, 2022-03-25:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**

**Hardware Version**

**Binary release or compiled yourself?**

**Operating System (and version)**

**Web Browser (and version)**

**Access Point**

**Describe the bug**

gabodiazjayme, 2022-03-25:
after flashing, the serial output shows "no SD card installer"

MartinMueller2003, 2022-03-25:
Which platform did you flash and please give a link to your hardware so I can see what the needed pin definitions are.

MartinMueller2003, 2022-04-02:
Does your hardware have an SD card slot and is there a card inserted into the slot? If not, then this is a normal message.

gabodiazjayme, 2022-04-04:
hello, i have added a reader module to the esp32, i have solved the problem by changing the SC from pin D5 to pin D4, now it works very well. thank you very much for your attention.