forkineye/ESPixelStick #513: Add support for ESP Web Tools based installation

henrygab, 2022-04-09:
**Is your feature request related to a problem? Please describe.**
While creating a "maker space" tutorial at work, a co-worker was trying to use ESPixelStick firmware.  The beta test / prototype run showed there was some friction caused by needing to install java (esp. on Mac).  In contrast, the WLED firmware provided simplified [installation](https://install.wled.me) using [ESP Web Tools](https://esphome.github.io/esp-web-tools/), which removed those hurdles / friction points.

**Describe the solution you'd like**
1. Support installation of ESPixelStick firmware using [ESP Web Tools](https://esphome.github.io/esp-web-tools/).
2. Also support IMPROV ([serial](https://www.improv-wifi.com/serial/)) to enable WiFi provisioning when using web installer.

**Additional context**
There's no better way to understand the benefit than to try the WLED installer (https://install.wled.me).

The website requirements for this are (from https://esphome.github.io/esp-web-tools/#add-website):
1. Having an HTTPS-enabled web site (a Web Serial security requirement)
2. Creation of [manifest](https://esphome.github.io/esp-web-tools/#manifest) file(s) for the various platforms
3. If the manifest files are fetched from a different server (than the web server), configuration of CORS-headers to allow those files to be fetched
4. Create webpage with user selection of platform, which defines which manifest to use.

**Open questions**
It appears that the current installer generates the littleFS file system dynamically, prior to upload.  This appears to be done to enable the following items:
1. Device ID and hostname to be provided from UI
2. Network SSID and password to be provided from UI

The network SSID and password are handled by IMPROV support (all happens locally at client viewing webpage.  However, I don't see IMPROV providing support for setting the deviceID and hostname. 

TO VERIFY:  Support for auto-generating a hostname/deviceID (e.g., based on MAC address) if no value exists in the configuration?

MartinMueller2003, 2022-04-10:
The newer version of the flashtool adds the ability to download an existing configuration and merge it with a new version of the software. It also generates log files as well as allows the rest of the network settings to be specified.

henrygab, 2022-04-13:
OK.  Web Serial Likely will not reach feature parity with the installer tool.  I still think it would add value.

If you also think this would add value, please leave this enhancement request open.

MartinMueller2003, 2022-04-13:
I have not looked into it and the final decision is up to Shelby. It is an interesting concept but it means rewriting the tool in JavaScript and using some as yet to be determined packages to make things easier. The concept is kind of cool.

MartinMueller2003, 2024-05-07:
Video here on how to use the web flasher.
https://www.youtube.com/watch?v=olu29-oUXTo
And the url for the tool: https://espixelstickwebflasher.from-ct.com:5000/

closed 2024-05-07 by MartinMueller2003