forkineye/ESPixelStick #516: ESP32 Cam no serial output

galle-led, 2022-04-13:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
4.0 beta-4

**Hardware Version**
ESP32 Cam

**Binary release or compiled yourself?**
Binary

**Operating System (and version)**
Windows 10 Home

**Web Browser (and version)**
Chrome

**Access Point**
Wireless Router

**Describe the bug**
ESPixelStick flasher works perfectly with the Wemos D1 Mini. Using the flasher tool, the code uploads to the ESP32 CAM but nothing displays on the serial monitor. I looked at the active connections on my router and the ESP32 CAM is not listed. Have tried 3 different ESP32 CAM boards. 

Status monitor shows a successful upload:

Writing at 0x003e4000... (85 %)
Writing at 0x003e8000... (100 %)
Wrote 196608 bytes (99332 compressed) at 0x003d0000 in 2.4 seconds (effective 665.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Done

-= Programming Complete =-

At this point if I press the reset button on the ESP32 CAM the serial monitor displays this:

rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download

MartinMueller2003, 2022-04-13:
Please use this version of the tool. After uploading the image, press the save log button and then append the log file to this issue
[dist.zip](https://github.com/forkineye/ESPixelStick/files/8484637/dist.zip)
.

MartinMueller2003, 2022-05-10:
updated dist,zip
[dist.zip](https://github.com/forkineye/ESPixelStick/files/8661253/dist.zip)

MartinMueller2003, 2022-07-28:
Any update on this? I have been working with the CAM board and not having an issue nor are the people I have been working with.

galle-led, 2022-07-29:
Hi Martin!

Thanks for checking back on this, just tried it again today and its working
fine, problem must have been on my end.

Thank you
-Steve

closed 2022-07-29 by MartinMueller2003