forkineye/ESPixelStick #518: GPIO Input to Trigger Sequence or Effect

MrPaulAR, 2022-04-16:
I would like to use any unused GPIO pins on the esp32 or esp8266 as inputs.  Any "special" pins should be excluded so that an end user doesn't do something unexpected such as GPIO0.

A basic HTML table input would be nice where these pins would be listed as a dropdown, one per row.  The dropdown would be preferred vs an input text box as it prevents user error.  One suggestion would be something like this.

| Enable | GPIO | Type | Sequence | Repeat  |
| ---- | ---- | ---- | ----  | ----  |
| Checkbox | Dropdown | Dropdown | Dropdown | Checkbox |

The "Type" field would be pullup or pulldown.  

There are two ways that could be used to draw this table.
* Default with one row that is not populated.  Have an "add" button that'll add another row.
* Since all common unused & "safe" GPIO pins are known it may be easier to just draw the entire table up front.

I also think there may be some other useful commands that could be triggered via GPIO inputs but more thought would be needed around that.
* Publish MQTT message
* HTTP command (GET, PUT, POST, etc)
* FPP remote command via unicast or multicast

MrPaulAR, 2022-04-16:
From https://github.com/forkineye/ESPixelStick/discussions/517

What happens at the end of the song? Does it stop? Play again? 
* User configurable as defined in the table above in the "Repeat" column.

What happens if a song is playing and the same button is pressed again? Does it ignore the press? Does it restart the song? 
* Suggestion is that the button press is ignored if less than 5 seconds.  If greater than or equal to 5 seconds the perform the action of the button.  This would account for both debounce and give the user control.

What if a song is playing and a different button is pressed? Do you move to the next song or remember the request and play the selected song next? How deep would the queue of songs be?
* Suggestion is that the button press is ignored if less than 5 seconds.  If greater than or equal to 5 seconds the perform the action of the button.  This would account for both debounce and give the user control.

MartinMueller2003, 2023-06-06:
A partial implementation has been added that triggers playing files or effects. It is not as versatile as this request so I do not consider this request to be fulfilled.