forkineye/ESPixelStick #526: EspixelStick Firmware -effect

brmeadows2841, 2022-05-03:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

No

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

I would like a means to edit or add effects to the ESP image. A user library of sorts. I know FSEQs can be uploaded to ESPs with an SD card, and set to play on startup or idle, but is it possible to add user effects to the ESP image? 

**Additional context**
Add any other context or screenshots about the feature request here.

brmeadows2841, 2022-05-04:
I see the built-in effects are in inputeffectsengine.hpp/cpp files but is there a method for an average user to add to the effects?

MartinMueller2003, 2022-05-04:
By modifying the code - Yes. By downloading something via the UI? - No. This would need to be considered. However, if you create your effects in a sequencer and upload the resulting fseq files you CAN download a playlist and cycle through any order of fseq files for any duration that you desire.

brmeadows2841, 2022-05-07:
How would a non coder bring in FastLed examples to Espixelstick image?

MartinMueller2003, 2022-05-07:
Replicate them in a sequencer, upload them and play them. Or put in a request to have them implemented in the ESP. Keep in mind that at the moment, the intent of this software is to help out in a show and provide a level of testing that can be used to find issues in the show configuration.

brmeadows2841, 2022-05-07:
Am i missing something? How can a seq be uploaded to an ESP that doesn't have an sd card?

brmeadows2841, 2022-05-07:
I get that the seqs that are built into the espixelstick software are for testing but it would be nice to have a means to swap them out for some other ones. Is there a framework or a guide I could follow to implement my own seqs. As this would allow for customization of the seqs for running on start up or idle, without having to upload to an SD card. If not that is fine as I have ordered a few SD card reader for a few naked ESPs.  What format does the espixelstick support, I get it supports Vixen and Xlights, but are there others?

MartinMueller2003, 2022-05-07:
Am i missing something? How can a seq be uploaded to an ESP that doesn't have an sd card?

Easy. Use the version that does have an SD card. The ESPv3 hardware HAS an SD card.

MartinMueller2003, 2022-05-07:
The SD card supports the same format your computer uses for USB drives. It has nothing to do with Vixen or xLights. .fseq file format is a well known format that Vixen and xLights can generate. If you wish to add to the existing list of effects, then you need to write code for the effects engine.

MartinMueller2003, 2022-05-21:
So the issue seems pretty simple. All of the D1 Mini implementations are using GPIO5 as the power control for the ethernet interface. The ESPv3 base card uses GPIO5 as the SD card Chip Select. A new base card is most likely needed that uses a different pin for the SD card.

MartinMueller2003, 2023-06-16:
FYI: The internal effects are NOT fseq files and do not use any sort of general purpose scripting.