forkineye/ESPixelStick #527: device setup, save change cannot be clicked

ddijanic, 2022-05-05:
I'm so sorry to bother you @MartinMueller2003 , but I have a problem again.

I bought
https://www.aliexpress.com/item/4000090521976.html?spm=a2g0o.order_list.0.0.155d1802qgsoJj

the d1_mini_mhetesp32minikit is the only one that works, the others have psram error. of course: D: D

the d1_mini_mhetesp32minikit works, but I can't save anything on device output ..
here is the log list ...

[ESP] ESPixelStick v4.x-dev (Mar 17 2022 - 09:56:19)
[ESP] v4.4-beta1-308-gf3e0c8bc41
[FileMgr] Flash file system initialized. Used = 204800 out of 327680
[FileMgr] Configuration File: '/config.json' loaded.
[FileMgr] Configuration File: '/output_config.json' loaded.
[OutputMgr] Invalid Channel Type in config '7'. Specified for channel '0'. Disabling channel
[OutputMgr] Invalid Channel Type in config '7'. Specified for channel '1'. Disabling channel
[FileMgr] Configuration File: '/input_config.json' loaded.
[InputMgr] Invalid Channel Type in config '7'. Specified for channel '0'. Disabling channel
[InputMgr] Invalid Channel Type in config '7'. Specified for channel '1'. Disabling channel
[FileMgr] Configuration File: '/config.json' saved 522 bytes.
[WiFiDrv] WiFi Entering State: Connecting Using Config Credentials
[WiFiDrv] Using DHCP
[WiFiDrv] Connecting to 'Dijanic stan' as esps-28CF2655B594
[WiFiDrv] WiFi Entering State: Connected To AP
[WiFiDrv] Connected with IP: 192.168.1.5
[WebMgr] Web server listening on port 80
[FPPD] Listening on port 32320
[EthDrv] Entering State: Powering Up
[EthDrv] Entering State: Connecting
E (13422) esp.emac: emac_esp32_init (355): reset timeout
E (13423) esp_eth: esp_eth_driver_install (222): init mac failed
[13430] [E] [ETH_m.cpp: 310] begin (): esp_eth_driver_install failed
[EthDrv] Entering State: Device Init Failed
[28384] [E] [vfs_api.cpp: 102] open (): /littlefs/www/artnet.html does not exist, no permits for creation
[WebMgr] WS client connect - 1
[WebMgr] WS client disconnect - 1
[72572] [E] [vfs_api.cpp: 102] open (): /littlefs/www/favicon.ico.gz does not exist, no permits for creation
[WebMgr] WS client connect - 2
[WebMgr] WS client disconnect - 2
[94445] [E] [vfs_api.cpp: 102] open (): /littlefs/www/favicon.ico.gz does not exist, no permits for creation
[WebMgr] WS client connect - 3

and DXM out can only on one output, not me on two, output 1 and output 2 would like or 3.
just offers me on OUTPUT1 DMX, Serial, Renard

my uload log :

https://anotepad.com/notes/bqe5e5n4

https://prnt.sc/qgOqmvCla9Sd
https://prnt.sc/Q-iqxL3_6m30

_Originally posted by @ddijanic in https://github.com/forkineye/ESPixelStick/discussions/502#discussioncomment-2688587_

MartinMueller2003, 2022-05-05:
I fixed that issue a few weeks back. It is caused by bad default values for some of the GPIO settings. What version are you using? Here is a dist installer for a more recent build.

[dist.zip](https://github.com/forkineye/ESPixelStick/files/8635630/dist.zip)

ddijanic, 2022-05-05:
you sent me dist.zip on [Mar 18]
now whit this one Save change works.
but ther is NO serial,DMX on OUTPUT 1
and output3 is reley onley option.

i need dual dmx out. how can i have thet ?

you can just tell me which board to buy from aliexpress ..
I found this now this has a psram https://www.waveshare.com/nodemcu-esp-s3-12k-kit.htm
or esp32 ttgo t8?

MartinMueller2003, 2022-05-09:
I just finished up the 10 port DMX changes. They will be in main in a few days.

The TTGO-T8 will work and has a built in SD card.

I just expanded the MHET minikit output ports a bit. You need to use the installer in the DIST.zip file or some other mechanism that clears the previous configuration. The error messages above are caused by a config mis-match.

MartinMueller2003, 2022-05-09:
I just added the DevkitC to the platforms list. A new dist file will be available shortly.

MartinMueller2003, 2022-05-09:
New dist.zip with devkitc support
[dist.zip](https://github.com/forkineye/ESPixelStick/files/8653218/dist.zip)

ddijanic, 2022-05-10:
I think he's wrong, I'll play with this tomorrow.
https://prnt.sc/GJgOfodu5iCo
Connecting to 'MaRtInG' as esps-28CF2655B594
trying to connect to MaRtInG wifi, and I didn't type that anywhere, and conf.json is empty.
and nepali AP mod.

MartinMueller2003, 2022-05-11:
Since you are using an image created by me for testing, you get to see my test network. MaRtInG is my isolated G-Band network used to test new code. It will revert to that network when all else fails. That gets removed when we do an official build.

ddijanic, 2022-05-11:
thanks for adding d1 d1 devkic...
https://prnt.sc/OOtvIv0_vviP

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12784
load:0x40080400,len:3032
entry 0x400805e4

[    ESP] ESPixelStick v4.x-dev (May  9 2022 - 11:39:38)
[    ESP] v4.4.1-1-gb8050b365e
[FileMgr] Flash file system initialized. Used = 217088 out of 327680
[   874][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (3) The physical drive cannot work      
[  1375][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[FileMgr] No SD card installed
[FileMgr] Configuration File: '/config.json' loaded.
[FileMgr] Configuration File: '/input_config.json' loaded.
[FileMgr] Configuration File: '/output_config.json' loaded.
[FileMgr] Configuration File: '/config.json' saved 356 bytes.
[WiFiDrv] WiFi Entering State: Connecting Using Config Credentials
[WiFiDrv] ERROR: STATIC SELECTED WITHOUT IP. Using DHCP assigned address
[WiFiDrv] Connecting to 'Dijanic stan' as esps-28CF2655
[WiFiDrv] WiFi Entering State: Connected To AP
[WiFiDrv] Connected with IP: 192.168.1.13
[ WebMgr] Web server listening on port 80
[   FPPD] Listening on port 32320
[ 13226][E][vfs_api.cpp:104] open(): /littlefs/www/index.html does not exist, no permits for creation  
[ WebMgr] WS client connect - 1
[FileMgr] ***Configuration File: '/output_config.json' too large for buffer. ***

MartinMueller2003, 2022-05-11:
Interesting, Too many features are enabled. I dont have one of those cards so I cant test for things like this.

ddijanic, 2022-05-12:
Maybe then raketa the d1 devkic build.
On monday I will order the TTGO-T8.
I hope thet will fix efriting and finaly have espixstick 🙂🙂🙂

MartinMueller2003, 2022-07-29:
Changes are in main

closed 2022-07-29 by MartinMueller2003