forkineye/ESPixelStick #530: FPP Remote Drif

sinh2000, 2022-05-15:
Thanks for all your great work!  I have firmware 4.0 beta 4 installed.  I am using a Raspberry Pi 4 as an FPP Player and have the ESPixelStick set as FPP Remote.  I have a 60-second sequence that I am playing in repeat mode.  At the beginning of the sequence each time, it is in sync.  It gradually gets worse and worse until by the end of the minute it's out of sync by a third to a half a second roughly.  I have recreated the sequence to make sure there wasn't something wrong with it.  Any ideas?  Thanks!

MartinMueller2003, 2022-05-15:
Try ome of the newer builds. a lot of chamges have been made since beta4

sinh2000, 2022-05-16:
Thanks!  I tried the latest build and it's working much better.  Appreciate it!

closed 2022-05-20 by MartinMueller2003