forkineye/ESPixelStick #534: FSEQ files incorrectly treated as playlist

henrygab, 2022-05-29:
https://github.com/forkineye/ESPixelStick/blob/7e9168e05ff9e58662248c3f69038e41c810c269/ESPixelStick/src/input/InputFPPRemote.cpp#L273-L275

At line 273, it checks the filename to see if ".pl" exists within, and if so, treats the file as a playlist.

This will correctly match the following types of filenames:
* `foo.pl`
* `bar.pl`
* `ReallyLongName.pl`

This will incorrectly match the following example filenames:
* `silver.plated.fseq`
* `sugar.plum.fseq`

If the extension for all playlist files is `.pl`, then the fix is simple ... check not only that the character sequence `.pl` exists, but that it's the full extension of the file.
e.g., the filename has more than three characters *AND* the last three characters are `.pl`.

MartinMueller2003, 2022-05-29:
Fixed in upcoming version. Also added checks for .fseq rather than assuming the file name was properly formatted.

MartinMueller2003, 2022-07-29:
Changes are in main

closed 2022-07-29 by MartinMueller2003