forkineye/ESPixelStick #536: Use the SD Card interface in 1bit mode.

Miata2k, 2022-05-31:
I have purchased a few of the following boards:
[https://www.olimex.com/Products/IoT/ESP32/ESP32-GATEWAY/open-source-hardware](https://www.olimex.com/Products/IoT/ESP32/ESP32-GATEWAY/open-source-hardware)
from Digikey for ~$20 each
[https://www.digikey.com/en/products/detail/olimex-ltd/ESP32-GATEWAY/8120280](https://www.digikey.com/en/products/detail/olimex-ltd/ESP32-GATEWAY/8120280)

The boards have an ESP32-Wroom-32E, Ethernet, SSD and a few I/O pins.

The SD card is wired for 1-bit mode as mentioned on this page:
[https://github.com/OLIMEX/ESP32-GATEWAY/tree/master/HARDWARE](https://github.com/OLIMEX/ESP32-GATEWAY/tree/master/HARDWARE)
"SD card changed from 4-bit mode of operation to 1-bit mode of operation"

Wiring Diagram.
[https://github.com/OLIMEX/ESP32-GATEWAY/blob/master/HARDWARE/Hardware%20revision%20G/ESP32-GATEWAY_Rev_G.pdf](https://github.com/OLIMEX/ESP32-GATEWAY/blob/master/HARDWARE/Hardware%20revision%20G/ESP32-GATEWAY_Rev_G.pdf)

I've been able to use the board with the Ethernet adapter on the latest CI build (ESPixelStick_Firmware-4.0-ci2406075826)

Would it be possible to add in (or turn on) the 1-bit sd interface option?   I see the 4 bit interface (SUPPORT_SD_MMC) option is not built in by default.  If the 1-Bit interface is enabled, would it be fast enough to stream a FSEQ file.

Board layout and I/O:
https://www.olimex.com/Products/IoT/ESP32/ESP32-GATEWAY/resources/ESP32-GATEWAY-GPIOs-Rev.F-up.pdf

If someone needs a board for testing I would be happy to supply one.

MartinMueller2003, 2022-05-31:
At this time, only 1 bit mode is available and has way more bandwidth than is needed to drive 3000 pixels.

Miata2k, 2022-05-31:
@MartinMueller2003 -  I'm sorry, I'm still a little confused.   Can you tell me which build is using the 1-bit interface with the SD card.   I've tested the SD card over SPI using the MISO MOSI interface by configuring the pins through the web interface, but the board I'm trying to use is not built for the SPI interface for the SD Card.

My questioning is based upon another website that indicate that power and ground are used for both interfaces, but SPI uses 4 lines back to the esp32, and 1-bit only uses 3 lines (15 to CMD, 14 to CLK, and 2 to D0).  Maybe the site is incorrect and I'm chasing the wrong problem.
https://www.instructables.com/Select-SD-Interface-for-ESP32/

I've reviewed the header files for all of the devices and they all seem to include an #define SUPPORT_SD (where possible) and then go on to define the SPI pins.  Only the ESP32_CAM has mapping to use the 4 bit interface, but again it doesn't seem to be enabled.

```
#define SUPPORT_SD
#ifdef SUPPORT_SD_MMC
#   define SD_CARD_CMD             gpio_num_t::GPIO_NUM_15
#   define SD_CARD_CLK             gpio_num_t::GPIO_NUM_14
#   define SD_CARD_DATA_0          gpio_num_t::GPIO_NUM_2
#   define SD_CARD_DATA_1          gpio_num_t::GPIO_NUM_4
#   define SD_CARD_DATA_2          gpio_num_t::GPIO_NUM_12
#   define SD_CARD_DATA_3          gpio_num_t::GPIO_NUM_13
#else // SUPPORT_SD
#   define SD_CARD_MISO_PIN        gpio_num_t::GPIO_NUM_2
#   define SD_CARD_MOSI_PIN        gpio_num_t::GPIO_NUM_15
#   define SD_CARD_CLK_PIN         gpio_num_t::GPIO_NUM_14
#   define SD_CARD_CS_PIN          gpio_num_t::GPIO_NUM_13
#endif // SUPPORT SD
```
The FileMgr.cpp makes reference to using the MMC interface but tries to start the SD card in 4 bit mode by using the ESP_SD.begin.
```
#ifdef SUPPORT_SD_MMC
        // DEBUG_V (String ("  Data 0: ") + String (SD_CARD_DATA_0));
        // DEBUG_V (String ("  Data 1: ") + String (SD_CARD_DATA_1));
        // DEBUG_V (String ("  Data 2: ") + String (SD_CARD_DATA_2));
        // DEBUG_V (String ("  Data 3: ") + String (SD_CARD_DATA_3));

        pinMode(SD_CARD_DATA_0, PULLUP);
        pinMode(SD_CARD_DATA_1, PULLUP);
        pinMode(SD_CARD_DATA_2, PULLUP);
        pinMode(SD_CARD_DATA_3, PULLUP);

        if(!ESP_SD.begin())
#else // ! SUPPORT_SD_MMC
```
SD_MMC.h show the option to add a flag for "mode1bit".

`bool begin(const char * mountpoint="/sdcard", bool mode1bit=false, bool format_if_mount_failed=false, int sdmmc_frequency=BOARD_MAX_SDMMC_FREQ);`

Would it be possible to define "SUPPORT_SD_MMC", change the pin modes for pints 2,14,and15, and then just call ESP_SC.being (mountpoint,true)  to mount the card using 1 bit mode?

From this site, this is what they are using for the ESP32_CAM module.
(https://dr-mntn.net/2021/02/using-the-sd-card-in-1-bit-mode-on-the-esp32-cam-from-ai-thinker)

I will test this configuration if I can ever get my development environment up and running on my Linux box.  I'm close, but that doesn't count in programming.  😄

MartinMueller2003, 2022-06-01:
Yes you could go to MMC 1 bit mode and yes it does have enough bandwidth or you could add the missing resistors back in and have the full MMC interface. FYI: MMC (1 or 4 bit) support is NOT tested. It is a work in progress, which is why it is not enabled in the ESP CAM implementation.

Miata2k, 2022-06-02:
@MartinMueller2003 

I made the changes as I tried to describe above, starting with the ESP32_QUINLED_QUAD_ETH configuration.

```
#define SUPPORT_SD_MMC
#ifdef SUPPORT_SD_MMC
#define SD_CARD_CMD             gpio_num_t::GPIO_NUM_15
#define SD_CARD_CLK             gpio_num_t::GPIO_NUM_14
#define SD_CARD_DATA_0          gpio_num_t::GPIO_NUM_2

//#   define SD_CARD_DATA_1          gpio_num_t::GPIO_NUM_4
//#   define SD_CARD_DATA_2          gpio_num_t::GPIO_NUM_12
//#   define SD_CARD_DATA_3          gpio_num_t::GPIO_NUM_13

//#else // SUPPORT_SD
#define SD_CARD_MISO_PIN        gpio_num_t::GPIO_NUM_19
#define SD_CARD_MOSI_PIN        gpio_num_t::GPIO_NUM_23
#define SD_CARD_CLK_PIN         gpio_num_t::GPIO_NUM_18
#define SD_CARD_CS_PIN          gpio_num_t::GPIO_NUM_5
#endif
```
```
#ifdef SUPPORT_SD_MMC
        // DEBUG_V (String ("  Data 0: ") + String (SD_CARD_DATA_0));
        // DEBUG_V (String ("  Data 1: ") + String (SD_CARD_DATA_1));
        // DEBUG_V (String ("  Data 2: ") + String (SD_CARD_DATA_2));
        // DEBUG_V (String ("  Data 3: ") + String (SD_CARD_DATA_3));

        pinMode(SD_CARD_DATA_0, PULLUP);
        //pinMode(SD_CARD_DATA_1, PULLUP);
        //pinMode(SD_CARD_DATA_2, PULLUP);
        //pinMode(SD_CARD_DATA_3, PULLUP);

        if(!ESP_SD.begin("/sdcard",true))
        
#else // ! SUPPORT_SD_MMC

```
This brought up the File Management tab on the UI and files on the card were visible.   I've uploaded a few others, but I'll need to do some more testing. 

Maybe this 1-bit mode could be used to fix the issues with the ESP32_CAM build.  The dr-mntn link suggest that the GPIO4 is also connected to an LED on the board.  Maybe that is the reason there are issues with its stability?