forkineye/ESPixelStick #543: PCA9685: change default servo position

zakkwillow, 2022-06-30:
I am currently using version 4.0 beta4  to control servos over wifi from xLights. I have noticed that when xLights is not outputting data the servos default to the lowest setting in the config tab. 

It would be great to have a selectable default position between min and max pulse width.

For example, I am building some animated skeleton props with 7 servos each. 2DOF for each arm and neck movement and a small jaw servo. Everything works great until the xLights sequence ends and then all the servos go to minimum position. This causes the prop to violently swing into an undesired position and quickly overheats my servos. I have built my props with the servos being centered as a neutral position. 

I realize that version 4 isn't finalized yet, but I feel that this would be a great addition for those of us working with servos. 

Thank you!
Zach

MartinMueller2003, 2022-06-30:
this will not be hard to add. Glad you are able to use the PWM outputs :)

zakkwillow, 2022-07-01:
You are awesome! I was struggling trying to figure out how to work around it. Now hopefully soon I won't have to. Many many thanks!

MartinMueller2003, 2022-07-01:
Changes to support this functionality are on my github repository and a pull request has been generated to transfer the changes to forkineye main.

MartinMueller2003, 2022-07-01:
@zakkwillow If you send me your email address to MartinMueller2003@yahoo.com, I will send you a link to my copy of the latest builds.

zakkwillow, 2022-07-01:
@MartinMueller2003 just got home from work and flashed your build. I can get it flashed and see in the output that it's connected to my network. My problem is when I try to access the settings from the browser. After I type in the IP address my browser tries to download a file with a name made from seemingly random characters and does nothing from there. 
I'm sure I'm doing something wrong, but I can't figure out what it is. Any ideas?

MartinMueller2003, 2022-07-02:
Connect the device to the USB cable
Flash the image (again)
Leave the device connected to the flash tool
Enter the ESP IP address in the browser and let it fail
On the flash tool click the save log button
Attach the resulting file to this message chain.

MartinMueller2003, 2022-07-02:
I think I know part of the problem. I forgot to put the updated web pages into the dist. New version has been placed in google.

zakkwillow, 2022-07-02:
Good morning, I just downloaded the new dist and flashed it. Same issues except now its not trying to download anything?
I've attached the log file.
[LogDump.zip](https://github.com/forkineye/ESPixelStick/files/9033207/LogDump.zip)

zakkwillow, 2022-07-02:
Also, if it helps, I'm using a wemos D1 mini (china clone ESP8266MOD) I also have ESP32 DevkitV1 boards for testing if needed

MartinMueller2003, 2022-07-02:
Yes knowing the hardware helps a lot.
That is bizarre. It was in a loop reading the config from littlefs until it ran out of memory, crashed and then booted up just fine.
And you say that after it boots, no pages get loaded. I can see that the Open Websocket message never arrives. I have not tested an ESP8266 build in a while. Can you try an ESP32  build just to see if it comes up? The TTGO build supports I2S output (that is what I used to test).

zakkwillow, 2022-07-02:
Just tried it with a fresh ESP32. Here's the log. I wish I knew more about the coding side of things so I could help more
[GenericESP32LogDump.zip](https://github.com/forkineye/ESPixelStick/files/9033449/GenericESP32LogDump.zip)
.

MartinMueller2003, 2022-07-02:
That error makes a bit more sense. You are trying to load an image that expects PSRAM to be present and when the device looked for the PSRAM, it was not there. While PSRAM is NOT mandatory, the system needs to know at compile time if it is there or not. Having PSRAM increases the number of pixels supported by the ESP.

Try loading the wt32eth01 image. It does not have PSRAM and it does have I2C on pins 21 (SDA) & 22 (SCL). It also has ethernet support but that wont hurt anything for your testing. You will get occasional messages indicating Ethernet failed to start.

zakkwillow, 2022-07-02:
Progress! I can pull it up in the browser now! It's only listing 2 channels for the PCA9685 though. Is that just a limitation for testing? 
I will need to order some more PCA9685s. I only have one and its already built into a project. Do you have a ESP8266 available to test with? I'd really like to stick with those because I have so many of them lol...

MartinMueller2003, 2022-07-02:
It should have listed all 16 channels. 
Yes I have an ESPixelstick V3 which has I2C turned off. I will be doing a custom build with it turned on. I wont have a lot more time to work on this until Tuesday.

MartinMueller2003, 2022-07-02:
Confirmed. The latest version of the ESP8266 code does not bring up the Browser page.

MartinMueller2003, 2022-07-02:
Issue #545 has been created to track this problem

zakkwillow, 2022-07-02:
Thanks again! Looking forward to updates!

MartinMueller2003, 2022-07-04:
Root cause is a lack of RAM. Increased the available ram by decreasing the number of concurrent pixel protocols supported by the configuration buffer. This frees up enough ram for the browser to connect. 

Uploaded a new dist.zip. The generic esp8266 build has I2C turned on and boots on an ESP8266 D1 Mini board.

zakkwillow, 2022-07-04:
Great, I'll get it set up and testing soon.

zakkwillow, 2022-07-05:
I got it flashed and it's exactly what I had asked for! I still only see 7 outputs for the PCA9685 which works fine for my application. I did experience drop outs of the webserver and loss of configured values however. I'm guessing this is part of the lack of ram problem?

Unfortunately I think the root of my issue lies with xLights. If I have output turned on after a sequence ends then xLights outputs a low servo value as well. 
If I turn output off, the servos go to the default value I've set on the hardware. Looks like I need to go bother the people at xLights now lol.

When I get some more time I will wire up to an ESP32 and see how that goes. 

Great work man, I really appreciate you taking the time. Let me know if there's anything you'd like me to try during my testing.

zakkwillow, 2022-07-06:
I confirmed it with xLights. The servo output defaults to zero when there is no information in the timeline. Similar to how the previous build of espixelstick did it. Seems like the only way around it according to the guys at xLights is to either turn the PWM output off when a zero is sent or cut power to the servo entirely.

zakkwillow, 2022-07-06:
That sounds like it would work. It's easy enough to set the servo limits in xLights. I'm game to try it if you are?

MartinMueller2003, 2022-07-06:
New version has been uploaded to my dist.zip (aka Alpha build)

zakkwillow, 2022-07-19:
Just wanted to give you an update. Ive been using your build and doing some programming. It is working well although I do get a lot of dropouts from the browser. I just ordered some Wt32-eth01 boards. I should get them tomorrow. Is there anything specific you'd like me to test out?

zakkwillow, 2022-07-19:
Also, thought you might get a kick out of my progress so far.

https://youtu.be/qJwwVR58PKY

MartinMueller2003, 2022-07-19:
Return those Eth01 boards as fast as you can. While I have one, it is the most unreliable board in the set. If you need Ethernet, grab one of the Quinled boards with Ethernet and SD support. They are rock solid.

MartinMueller2003, 2022-07-19:
That is really cool. What kind of laser are you using?

zakkwillow, 2022-07-21:
It's a 2w RGB laser I got from AliExpress. Emma brand. Controlling with Helios and Lasershowgen.

MartinMueller2003, 2022-07-29:
Changes are in main

closed 2022-07-29 by MartinMueller2003