forkineye/ESPixelStick #549: FPP remote not outputting to lights - ESP32cam 4.0beta-4

Mark79864, 2022-07-17:
- ESP32cam with 4.0beta-4.
- Binary release and flashed with included tool.
- Windows 10, Chrome browser.
- Ubiquiti AP PRO access point, -46dBm / 100%. Also tried on a standard ISP router's access point, -53dBm / 94%. IP: 192.168.1.23
- Output 1, GPIO-0 (default), WS2811, 10 GRB pixels. 
- xSchedule 2022.14 64bit.

[image]

What is working:
Primary input of E1.31 and DDP are working from xLights over WiFi to the ESP32cam.
Secondary input of 'FPP remote' and selecting a specific FSEQ is also working.
[image]

**Bug:**
FPP remote does not output to the lights.
xSchedule was used to send FPP sync data. It has been tried in Multisync, Unicast and Broadcast.
Home page shows that ESpixelstick is receiving sync data and is correctly showing the elapsed/remaining time within 3s. But there is no output to the lights.
[image]

MartinMueller2003, 2022-07-17:
I have attached the latest build dist to this response. Please load the appropriate image and (leaving the installer turned on) try the failure case. Then click the save to log button at the bottom of the installer and attach the log to this ticket.

MartinMueller2003, 2022-07-17:
It looks  like Git no longer allows me to upload the file. Send me your email and I will give you access to the version on googledrive.

Mark79864, 2022-07-18:
> It looks like Git no longer allows me to upload the file. Send me your email and I will give you access to the version on googledrive.

If you are the same Martin as on ACL, I can private message you on there.
[https://auschristmaslighting.com/members/mark_m.29098/]

I will send an email to the associated address on your Github profile.

Mark79864, 2022-07-19:
> Were you able to get the latest dist.zip file?

Yes, I have got the dist.zip folder you have sent me.
I'll finally be free this afternoon to be able to trial it out. The week so far has been busy sorry.

Mark79864, 2022-07-20:
Okay, I have got onto trying again.
In the weekend I will try further to see what is happening.

In the past two hours:
- Flashed the firmware you sent me successfully.
- ESpixelstick web interface came up.
- Issue: SD card was not recognised. After unplugging it, rebooting, etc, it still didn't show up.
- Then a factory reset to see if that helps with the SD card.
- The SD card was then recognised however I couldn't get it back onto my main WiFi network from the hotspot...

Odd behaviour. I shall test further because this seems far too odd to be part of this bug.

Side note:
The ESP32Camera's LED lights is set to be constantly on with 4.0beta-4 on Github and the version you sent me.
It is insanely bright.

Mark79864, 2022-07-20:
Ah crap. My finger hit the close issue lol.
Comment to reopen...

MartinMueller2003, 2022-07-20:
The factory reset discarded your WiFi settings. That is why it stayed off the network. It is interesting that the SD card was recognized after the factory reset. That means there is something wrong with the uploader. At any rate, the ESP will enter AP mode and you can connect to it using a tablet or smartphone. Then you can set the WiFi credentials and you will be back online.

MartinMueller2003, 2022-07-20:
If you look in the readme, you should find a description of how to put the WiFi credentials on the SD card. After a factory reset the ESP will look at the SD card and use those credentials to connect AND then delete the file.

Mark79864, 2022-07-20:
> The factory reset discarded your WiFi settings. That is why it stayed off the network. It is interesting that the SD card was recognized after the factory reset. That means there is something wrong [...]

I don't know whether it was my error or not for connecting back to my main WiFi network. I used the hot spot to enter the SSID/Password but it didn't seem to want to connect.

> That means there is something wrong with the uploader. 

Interesting you mention this.
Could the issue of flashing be the cause of FPP remote not working....?

More to follow in the weekend:

Mark79864, 2022-07-27:
For completeness on Github, Martin and I have talked elsewhere over this.
With the code that Martin has sent, FPP remote is now working fine :).

closed 2022-07-27 by Mark79864