forkineye/ESPixelStick #555: Error compiling SOURCE

chassy33, 2022-08-30:
I am attempting to compile and build from source but I keep getting an error that just says `Error:`. The error occurs when attempting to install espressif/toolchain-xtensa-esp32. I am new to building from source, but I am just getting no verbose errors to show me what is going on. I followed the instructions in the readme but can't seem to get anywhere. 

[image]

MartinMueller2003, 2022-08-31:
That is a problem within platformio. I'll take a look but you would be best served by asking in a platformio forum. There may simply be a setting in PIO that did not get set up on your computer.
Martin

MartinMueller2003, 2022-09-01:
I think I may have tracked this down. There is an incompatibility in the WebServer library and the new ESPIDF. I got a similar error after deleting the .pio directory and forcing a complete new build. Please download the latest code, delete the .pio directory (If it is present) and do a build.

chassy33, 2022-09-01:
Martin,

Thank you for the quick replies! I have not had a chance to do any more testing. I did ask some questions in a platformIO group but got no real answers.  I will give that a try though and let you know.

MartinMueller2003, 2022-09-14:
Have you verified the fix? I would like to close this.

MartinMueller2003, 2022-09-29:
Still not seeing any more build errors

closed 2022-09-29 by MartinMueller2003