forkineye/ESPixelStick #559: FPP Master in master/remote configuration

brmeadows2841, 2022-09-01:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

Not related to a issue.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

I request a feature that would allow an ESP to act as a master FPP in master/remote configuration.

**Additional context**
Add any other context or screenshots about the feature request here.

"If you are asking for a new feature in which one of the ESP V4s pretends to be an FPP Master, yes that could be added. However, based on my schedule, that would be started in February. If you want this, please put in a feature request."

MartinMueller2003, 2022-10-05:
Are ou also asking that the ESP have a scheduler built in? Just sending synch packets when playing a file is not difficult. Setting up aa scheduler is a very different thing.

brmeadows2841, 2022-10-05:
Sending sync packets would be start.

Example, imagine a group of ESPs with one being the master player, the other ones in sync with the master.
The master is either playing a playlist or triggered by a GPIO event.

MartinMueller2003, 2023-06-16:
FYI: Having an FPP master connected to the network via WiFi will require the Wireless network to be super stable AND all of the remote devices will need to have a positive offset configured on the FPP slave devices. A WiFi FPP Master will result in TWO WiFi transactions. One from FPP Master to the AP and one from the AP to the FPP Slaves. This means there are two delays in sending the sync PDU and each of the delays may be changing.

With that said, I am going to add the feature but someone will have to test it on a complex network.

brmeadows2841, 2023-06-17:
I am willing to test it, I have a network with two different Kulp controllers and various ESPs (3 Raspberry pi 3, 3 PI4, 1 esp8266s, 3 esp32 cams, and other esp32s. Connections are wired and wireless, straight and through hubs and switches. My Wi-Fi setup is four different routers that cover 1 1/2 acres.

MartinMueller2003, 2023-06-17:
I got bored while on vacation and coded this up. I have no way to test anything so next steps will have to wait until early July. At the moment, I don't even know if the sync PDUs are set up properly.

brmeadows2841, 2023-06-17:
> I got bored while on vacation and coded this up. I have no way to test anything so next steps will have to wait until early July. At the moment, I don't even know if the sync PDUs are set up properly.

Was there an attachment?

MartinMueller2003, 2023-06-17:
Nope. I have everything pushed to my fork on github.

MartinMueller2003, 2024-01-01:
This has been on the mainline for a few months now. I would like to close this ticket.

closed 2024-01-06 by MartinMueller2003