forkineye/ESPixelStick #562: Python Error ESPSFashTool Mac 12.4

k5beason, 2022-09-15:
Hello,

When trying to use the ESPSFlashTool.jar on My M1 Mac running 12.4 I get "java.io.IOException: Cannot run program "python": error=2, No such file or directory" in the terminal and I'm unable to flash.

python --version returns Python 3.8.9

I've tried both ver 3.2 and ver 4.0 beta 4.

forkineye, 2022-09-15:
Have you tried running it from the command line? java -jar ESPSFlashTool.jar.

k5beason, 2022-09-15:
> Have you tried running it from the command line? java -jar ESPSFlashTool.jar.

Yes. Its the only way the flash tool will open. No errors until I go to flash and then it states unable to connect, and in terminal it indicates the python error

forkineye, 2022-09-15:
what does 'which python' return?

k5beason, 2022-09-15:
> what does 'which python' return?

kbeason@MacBook-Pro ~ % which python
python: aliased to /usr/bin/python3

forkineye, 2022-09-15:
Ah, you need to setup a symlink to python3 as the alias won't work from within java.  Like ```ln -s /usr/bin/python3 ~/bin/pyhton```  if you have a bin directory in your home and path. If not, ```sudo ln -s /usr/bin/phthon3 /usr/local/bin/pyhton``` should work.

k5beason, 2022-09-16:
> Ah, you need to setup a symlink to python3 as the alias won't work from within java. Like `ln -s /usr/bin/python3 ~/bin/python` if you have a bin directory in your home and path. If not, `sudo ln -s /usr/bin/python3 /usr/local/bin/python` should work.

Linking to the preinstalled python would trigger Xcode to install additional tools everytime the flash tool was run. Never worked even after installing the prompted tools.

I discovered I needed to download and install python3 from python.org and then used 
sudo ln -s /Library/Frameworks/Python.framework/Versions/3.10/bin/python3 /usr/local/bin/python

Both my M1 Macs have the same issue with python. Not sure if it can be addressed in future releases.