forkineye/ESPixelStick #566: Apple Silicon - ESPSFlashTool.jar wont work

ilu2k21, 2022-09-28:
Hi, 
the flash tool wont work here on my 2020 Mac MIni, starting from terminal i get following error:

Exception in thread "AWT-EventQueue-0" java.lang.UnsatisfiedLinkError: Can't load library: /var/folders/c6/1g5699210910vkzfsjnpccm40000gn/T/1664360507374-libjSerialComm.jnilib

Any hints?

forkineye, 2022-09-28:
Are you running it from within the top-level directory where it's extracted? ```java -jar ESPSFlashTool.jar```

forkineye, 2022-09-28:
Also, what does ```which python``` say?

ilu2k21, 2022-09-29:
Yes, started it from within the extracted directory.
which python says: /opt/homebrew/bin/python

MartinMueller2003, 2022-10-11:
First, use the version found at the link below until the next beta is released. Next, it looks like you have an incomplete version of Java installed. Make sure you have the latest Java on your machine. Also, your error looks similar to the issue described in this discussion:
https://stackoverflow.com/questions/71237976/unsatisfiedlinkerror-when-running-typedb-on-an-m1-macbook

dist.zip

| 
| 
|  | 
dist.zip

 |

 |

 |

gwolanin, 2022-12-07:
I'm using a 2020 M1 MacBook Pro and it works successfully for me.

java -version
java version "1.8.0_351"
Java(TM) SE Runtime Environment (build 1.8.0_351-b10)
Java HotSpot(TM) 64-Bit Server VM (build 25.351-b10, mixed mode)

java -jar ESPSFlashTool.jar
OS: mac os x / x86_64
Detected Mac

forkineye, 2022-12-07:
I've tested this on the latest MacOS release recently as well.  It needs to be ran from the command line where it's extracted - "java -jar ESPSFlashToo.jar". The first time running, you'll receive the typical MacOS security notifications requiring you to go into your security settings and allow mklittlefs to run.

closed 2022-12-07 by forkineye

mores, 2023-02-04:
I am seeing the same issue:

% uname -a
Darwin Stephens-Mini.localdomain 22.2.0 Darwin Kernel Version 22.2.0: Fri Nov 11 02:04:44 PST 2022; root:xnu-8792.61.2~4/RELEASE_ARM64_T8103 arm64

% java -version
openjdk version "17.0.1" 2021-10-19 LTS
OpenJDK Runtime Environment Zulu17.30+15-CA (build 17.0.1+12-LTS)
OpenJDK 64-Bit Server VM Zulu17.30+15-CA (build 17.0.1+12-LTS, mixed mode, sharing)

% java -jar ESPSFlashTool.jar
OS: mac os x / aarch64
Detected Mac
Exception in thread "AWT-EventQueue-0" java.lang.UnsatisfiedLinkError: /private/var/folders/c5/4jbz7nn91zbdkjb9sc2by5_w0000gn/T/1675517958973-libjSerialComm.jnilib: dlopen(/private/var/folders/c5/4jbz7nn91zbdkjb9sc2by5_w0000gn/T/1675517958973-libjSerialComm.jnilib, 0x0001): tried: '/private/var/folders/c5/4jbz7nn91zbdkjb9sc2by5_w0000gn/T/1675517958973-libjSerialComm.jnilib' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64')), '/System/Volumes/Preboot/Cryptexes/OS/private/var/folders/c5/4jbz7nn91zbdkjb9sc2by5_w0000gn/T/1675517958973-libjSerialComm.jnilib' (no such file), '/private/var/folders/c5/4jbz7nn91zbdkjb9sc2by5_w0000gn/T/1675517958973-libjSerialComm.jnilib' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64'))
	at java.base/jdk.internal.loader.NativeLibraries.load(Native Method)
	at java.base/jdk.internal.loader.NativeLibraries$NativeLibraryImpl.open(NativeLibraries.java:384)
	at java.base/jdk.internal.loader.NativeLibraries.loadLibrary(NativeLibraries.java:228)
	at java.base/jdk.internal.loader.NativeLibraries.loadLibrary(NativeLibraries.java:170)
	at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2389)
	at java.base/java.lang.Runtime.load0(Runtime.java:755)
	at java.base/java.lang.System.load(System.java:1953)
	at com.fazecast.jSerialComm.SerialPort.<clinit>(SerialPort.java:181)
	at com.forkineye.espsflashtool.ESPSFlashToolUI.<init>(ESPSFlashToolUI.java:235)
	at com.forkineye.espsflashtool.ESPSFlashToolUI$8.run(ESPSFlashToolUI.java:908)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:318)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:771)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:722)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:716)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:741)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)