forkineye/ESPixelStick #567: DMX not working as expected on RTM pin

processor83, 2022-10-14:
**ESPixelStick Firmware Version**
Curent version of master branch (4.x.x)

**Hardware Version**
Esp3DeuxQuattro+DMX
its a esp32-wroom based board with Lan8720 ethernet that use esp32 clock
DMX port use AM26C31 differential line driver

**Binary release or compiled yourself?**
Compiled myself with platform.io and binary provided by Martin Mueller

**Operating System (and version)**
Windows 10

**Web Browser (and version)**
Chrome

**Describe the bug**

When DMX output are set on RTM pins it partilay work as im able to turn channels on and off but they dont receive the 0-255 value as expected.   For exampel brightness of a channel will be into steps so let say 25/50/75% and then 25/50/75% again   so a fade from 0 to 255 will make the channel flash from low to high bringthness instead of a gradual ramp to 100%.

tested same pins hardware but defined the two dmc hardwrae outoput as UART and it work as expected.

@MartinMueller2003 told me thet DMX should work on RMT pins but that UARt is better.

processor83, 2022-10-14:
I did a test and juste changed uart to RTM changed nothing else and build this and it acting like the version that @MartinMueller2003 sensed to me couples weeks ago, did this to make sure it was not older bug that was fixed.

So on RTM ACT Very weird it send the 3 dmx channels for rgb in my test but it's the 0-255 value that is corrupted. How can I share a video here?

processor83, 2022-10-14:
This dont work:
#define DEFAULT_RMT_0_GPIO     gpio_num_t::GPIO_NUM_12
#define DEFAULT_RMT_1_GPIO     gpio_num_t::GPIO_NUM_2

This work:
#define DEFAULT_UART_1_GPIO     gpio_num_t::GPIO_NUM_12
#define DEFAULT_UART_2_GPIO     gpio_num_t::GPIO_NUM_2

processor83, 2022-10-14:
https://user-images.githubusercontent.com/35543827/195939184-c400bc40-c045-40e8-a761-472317f7f1ce.mp4

https://user-images.githubusercontent.com/35543827/195939199-c2fea84f-4278-476b-8f72-bb77033c6f16.mp4

MartinMueller2003, 2022-10-20:
I Have been looking a a bunch of output values between the various outputs and I do not see any difference between them. The only think I can think of is that the polarity may be reversed on the DMX outputs. I have attached a screenshot of a logic analyzer capture of the channels outputting data. 

[image]

MartinMueller2003, 2022-10-20:
FYI: My probes are on the D pins and A pins on the output connectors.

processor83, 2022-10-20:
with out any hardware change rmt does not work and uart work  thats why i dont Under stand.  Could it be the speed of the serial is different ?

are you able to check if there is a difference  on a normal port with DMX set so no differential line driver involved mays the difference happen at the input ???

what equipment i can get to do so testing on my side?

processor83, 2022-10-20:
Also what did you capture what event?? because it acted the same to turn it on and off at 100% for example but not at different % change.

MartinMueller2003, 2022-10-20:
I did 0, 255, 128. If you can give me a number that is a problem then I will test it.

The UARTs 1 and 2 are on Pixel ports 1 & 2, RMTs 0 & 1 are on pixel ports 3 & 4, RMTs 2 & 3 are on DMX ports 1 & 2. Those traces are all taken at the same moment in time. as I said, timing variation between them is within 1us.

processor83, 2022-10-20:
I did my test with only two port the gpio2 and 12 used for dmx port 1 and 2. And with those two port tested uart and rmt maybe the diff is here.
 I saw a logic analyser I ca build with a fr escale board I have so I will build one and test on my side. Will also try to give you two number that give same brightness but should not!

MartinMueller2003, 2022-10-20:
I got mine at Amazon for less than $50 USD

[image]

And here is a pic of the analyzer
[image]

https://www.amazon.com/gp/product/B077LSG5P2/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

MartinMueller2003, 2022-10-20:
[image]

MartinMueller2003, 2022-10-29:
Can you send me the values that are not working?

processor83, 2022-10-29:
Will give you exact example later tonight I will also flash and try the Logic analyzer on my Freescale board and will try to capture the problem!

processor83, 2022-11-01:
ok so for example on RMT  50/255 100/255 150/255 all look the same.  25,75,125/255 all look the same..  3/255 and 255/255 look the same.
When i use UART then it is gradual like it is supposed..

here what i camptured on my logic analyser (new to me i dont really know how to use it but that it thats when it sedn 100/255 backgroud in xlight test mode.

Channel-5 is UART
Channel-3 is RMT

[image]

processor83, 2022-11-01:
[image]

processor83, 2022-11-01:
i capture directly on the esp32 module 3.3v signal since my analyser is only 3.3v.

MartinMueller2003, 2022-11-01:
OK Thanks. I will try these values when I get home

processor83, 2022-11-01:
and it is the same problem if i use one slider to change only one channel ou for example here use test tool in xlight and chnage background slider.

MartinMueller2003, 2022-11-03:
I found the issue. Changes have been pushed to my branch. I will be updating the dist.zip in a few hours. I am real embarrassed by this one. The RMT driver was sending the data MSB to LSB as it is done for pixels. DMX sends data LSB to MSB. I added code to support that in the RMT driver. UART driver already had that supported which is why it worked for UART based ports and not RMT based ports.

MartinMueller2003, 2022-11-03:
UART output
[image]

RMT OUTPUT
[image]

MartinMueller2003, 2022-11-03:
new dist.zip
https://drive.google.com/file/d/1DG3wz8lbYNc371OIbeOqkzMaY5Zd2SKA/view?usp=sharing

processor83, 2022-11-05:
> new dist.zip https://drive.google.com/file/d/1DG3wz8lbYNc371OIbeOqkzMaY5Zd2SKA/view?usp=sharing

Just in case it contain your wifi credentials...

processor83, 2022-11-05:
> new dist.zip https://drive.google.com/file/d/1DG3wz8lbYNc371OIbeOqkzMaY5Zd2SKA/view?usp=sharing

I did test it ton my dmx dumb strip driver and it work perfectly !! as expected!

MartinMueller2003, 2022-12-10:
Changes are on main and in beta 4