forkineye/ESPixelStick #568: QuinLED-ESP32 or QuinLED-ESP32-AE+ Outputs

onewithhammer, 2022-10-20:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
4.0-ci3138152858

Flashed using ESPSFlashTool with DIG-QUAD selected as hardware.

**Hardware Version**
QuinLED-ESP32 or QuinLED-ESP32-AE+

**Binary release or compiled yourself?**
Release

**Describe the bug**
In earlier release (4.0-ci2756477305) the number of outputs was 6 outputs.  This current version only has 5 outputs.  Which is correct?

What about supporting the QuinLED-ESP32-AE+ which has 3 additional LED outputs on a DIG-QUAD?

Is this currently possible?

Please advise.

I thought the ESP-32 had a max of 6 outputs.
[image]

ravenridgelights-turtle, 2022-10-20:
There is a new distro for 4.x that has support for the ae+ Martin has put together.  Idk if I can post links here but you can get it in dr.zzs discord # led channel in the pins from Martin himself .. I'm not sure where else it's available

onewithhammer, 2022-10-20:
I found the Distro.zip on dr.zzs discord #led channel.  Is that the latest version for AE+?
4.0-ci1789220604

MartinMueller2003, 2022-10-20:
the link points to my latest compiled versions. I overwrite the old version of the zip file when a new set is created. So the dist.zip is always the most recent build.

ravenridgelights-turtle, 2022-10-20:
> @MartinMueller2003 <https://github.com/MartinMueller2003> - Thanks for
> the info on your link. I have already downloaded and flashed a
> QuinLED-ESP32-AE+ and a QuinLED-ESP32.
>
> The firmware - device config for QuinLED-ESP32 displays 5 outputs which is
> what I expect. 4 outputs and the 5th for relay.
>
> The firmware - device config for QuinLED-ESP32-AE+ displays 6 outputs. 5
> outputs and the 6th for relay. Shouldn't it display 8 outputs, 7 outputs
> and the 8th for relay? Am I missing something? How do you use all 7 LED
> channels on the AE+ ?

Mine most definitely displays 7led and one relay

ravenridgelights-turtle, 2022-10-20:
> @MartinMueller2003 - Thanks for the info on your link. I have already downloaded and flashed a QuinLED-ESP32-AE+ and a QuinLED-ESP32.
>
> The firmware - device config for QuinLED-ESP32 displays 5 [...]

https://youtu.be/GkR-xJrUhvg

onewithhammer, 2022-10-20:
@ravenridgelights-turtle - My mistake.  You are correct.  I was flashing with QUINLED UNO AE+ , NOT QUINLED QUAD AE+

onewithhammer, 2022-10-20:
Last question - xLights only shows 6 outputs for the DIG-QUAD even with AE+ ?

ravenridgelights-turtle, 2022-10-20:
That one I can't explain

MartinMueller2003, 2022-10-20:
I have no control over how xLights determines how many ports are available. The config file contains all of the information needed to figure that out (that's how the browser UI does it). You would need to ask the question on an xLights forum.

onewithhammer, 2022-10-20:
Thanks.  I will ask the xLights developers. 
This issue can be closed.

patdelaney, 2022-10-20:
Someone tell me what it should be and it can be added,  it's just config
file update. I'll send it to the developers.

What are the options that it should allow.

pat

brmeadows2841, 2022-10-21:
This shows  GPIOS pins 16,32,33,4,12,2 as the pixel control pins.
Can  GPIO pins 14,15 or 13 also be used, without conflicts? 

This would be 9 outputs on an ESP32.
I assume the pixel limit would still be ~3000 across all ports.

MartinMueller2003, 2022-10-21:
You can but I have found that the ESP32 starts having issues when more than 8 pixel ports are in use. 

FYI: The configuration above is a "Random" sampling of the configs that ESPixelStick V4 supports. At this time there are more than 20 supported hardware configs and the config is different for each of them. My UI code looks at the options in the file and uses that to create and populate the configuration choice lists. Nowhere is there the concept of "This platform type uses this fixed configuration".

brmeadows2841, 2022-10-21:
How do I change the UI to show the additional ports, from 6  to 8?
Where is this configuration file located, do I change it before using Visual Studio to implement the image?

> You can but I have found that the ESP32 starts having issues when more than 8 pixel ports are in use.

I will stay at eight for the maximum number of ports.
Is this a timing issue, ram of some other type of problem?

> FYI: The configuration above is a "Random" sampling of the configs that ESPixelStick V4 supports. At this time there are more than 20 supported hardware configs and the config is different for each [...]

MartinMueller2003, 2022-10-21:
You create a new platform definition in the platform directory, add it to the gpiodefs file. Add the new board type to the platformio.ini file. Do a build of your new platform and upload it to the ESP.

MartinMueller2003, 2022-10-22:
Take a look at the DIG-Octa platform. It has 8 ports. It does not have PSRAM which is something you really should have.