forkineye/ESPixelStick #573: Failed to Connect

ldismo, 2022-10-29:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
ESPixelStick_Firmware-3.2

**Hardware Version**
ESPixelStick v3.0

**Binary release or compiled yourself?
Just trying to load the basics

**Operating System (and version)**
Windows 11

**Web Browser (and version)**
Chrome

**Access Point**
Issue Loading from PC

**Describe the bug**
I'm trying to use the ESPixelStick Flash Tool to flash the latest version and configure it for my network.  I get this error. 
-= Building SPIFFS Image =-
 bin/win32/esp8266/mkspiffs -c spiffs/ -p 256 -b 4096 -s 131072 firmware/spiffs.bin
/config.json
/www/esps.css.gz
/www/esps.js.gz
/www/index.html.gz
Done

-= Programming ESP8266 =-
 bin/win32/python3/python bin/upload.py --chip auto --baud 115200 --port COM5 --before default_reset --after hard_reset write_flash 0x0 firmware/pixel-3.2-1m_128k.bin 0xDB000 firmware/spiffs.bin
esptool.py v2.8
Serial port COM5
Connecting........_____....._____....._____....._____....._____....._____....._____
Traceback (most recent call last):
  File "bin/upload.py", line 19, in <module>
    esptool.main(sys.argv)
  File "C:/Users/dismo/Downloads/ESPixelStick_Firmware-3.2/ESPixelStick_Firmware-3.2/bin/esptool\esptool.py", line 2883, in main
    esp = ESPLoader.detect_chip(each_port, initial_baud, args.before, args.trace)
  File "C:/Users/dismo/Downloads/ESPixelStick_Firmware-3.2/ESPixelStick_Firmware-3.2/bin/esptool\esptool.py", line 274, in detect_chip
    detect_port.connect(connect_mode)
  File "C:/Users/dismo/Downloads/ESPixelStick_Firmware-3.2/ESPixelStick_Firmware-3.2/bin/esptool\esptool.py", line 483, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to Espressif device: Timed out waiting for packet header
Done

MartinMueller2003, 2022-10-30:
One of two things are wrong: You are using the wrong com port or your platform does not support NodeMCU programming via the UART control pins and you need to put your device into programming mode manually.

forkineye, 2022-10-30:
Can you post a screenshot of the COM ports available in the drop down of the flash tool?

MartinMueller2003, 2022-10-31:
Please send a link to the device you purchased. There are many products that go by the name of "D1 Mini"

ldismo, 2022-10-31:
> When you press the reset button, do you see anything in the serial output
> window? Something could have happened to the Wemos. If you can, send me
> your order number.
> thanks,
> -shelby

Order# 112-6918992-9279414
There is no display on the D1 Mini.... do you mean on the ESPFlash Tool?

MartinMueller2003, 2022-10-31:
He is asking this:

1. Connect the USB cable from the PC to the ESP
2. Start up the flash tool
3. On the flash tool:
4. Set the port to the correct COM value
5. Set the Firmware selection to the proper value
6. On the ESP, press the reset button
7. On the flash tool console (lower window), you should see some output. Paste it to this message chain

What

ldismo, 2022-10-31:
Interesting, the string is odd.
bd cl�r{l cd�s� ��c� ��c ���;� l � �od�s� � 'og c �lc

[image: image.png]

ldismo, 2022-11-02:
> The attachment didn't come through, but it sounds like the module is
> alive. Does the string repeat each time you press the reset button?

The string is shown below and was also in the screenshot.  Is this normal?
bd cl�r{l cd�s� ��c� ��c ���;� l � �od�s� � 'og c �lc

[image: image.png]

forkineye, 2022-11-02:
Something may have happened to the Wemos. Have you tried another one, and does it do the same? Is it still flashed from before / can you access it via the web?

ldismo, 2022-11-02:
I tried another Wemos and was able to flash it.  Thanks... I guess the
first one is toast.

MartinMueller2003, 2022-11-04:
This looks to have been a hardware issue.

closed 2022-11-04 by MartinMueller2003

ldismo, 2022-11-04:
Ok, thanks.