forkineye/ESPixelStick #584: Cant connect to web user interface or get Ethernet IP adresse  without Wifi available

sonofice80, 2022-11-15:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
ESPixelStick_Firmware-4.0-ci3445467370

**Hardware Version**
Quinled-esp32

**Binary release or compiled yourself?**
binary

**Operating System (and version)**
Win10

**Web Browser (and version)**
Chrome

**Access Point**

**Describe the bug**
Cant connect to web user interface or get Ethernet IP adresse  without Wifi available
-Have tried with static and dhcp on both Wifi and ethernet connections
-Have tried with and without "allow WIFI"

Serial Output:

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12784
load:0x40080400,len:3032
entry 0x400805e4

[    ESP] ESPixelStick v4.0-ci3445467370 (Nov 11 2022 - 13:47:13)
[    ESP] v4.4.1-1-gb8050b365e
[FileMgr] Flash file system initialized. Used = 204800 out of 327680
[FileMgr] Configuration File: '/config.json' loaded.
[    ESP] Processing Flash Tool config
[NetworkMgr] Using old style WiFi Settings
[NetworkMgr] No network Ethernet settings found. Using default Ethernet Settings
[   427][E][vfs_api.cpp:104] open(): /littlefs/input_config.json does not exist, no permits for creation
[FileMgr] Configuration File: '/input_config.json' saved 934 bytes.
[   734][E][vfs_api.cpp:104] open(): /littlefs/output_config.json does not exist, no permits for creation
[FileMgr] Configuration File: '/output_config.json' saved 6981 bytes.
[FileMgr] Configuration File: '/input_config.json' loaded.
[FileMgr] Configuration File: '/output_config.json' loaded.
[FileMgr] Configuration File: '/config.json' saved 559 bytes.
[WiFiDrv] WiFi Entering State: Connecting Using Config Credentials
[WiFiDrv] Using DHCP
[WiFiDrv] Connecting to 'Lysbordet' as Fly
[WiFiDrv] WiFi Entering State: Connected To AP
[WiFiDrv] Connected with IP: 192.168.0.38
[ WebMgr] Web server listening on port 80
[   FPPD] Listening on port 32320
[ EthDrv] Entering State: Powering Up
[ EthDrv] Entering State: Connecting
[ EthDrv] ETH IP Before Start: 0.0.0.0
[ EthDrv] Ethernet Connecting as Fly
[ EthDrv] Connecting to Ethernet using DHCP
[ EthDrv] Entering State: Wait for IP Address
[ EthDrv] Entering State: Got IP
[ EthDrv] Ethernet Connected with IP: 192.168.0.41
[ EthDrv] Entering State: Connecting
[ EthDrv] Connecting to Ethernet using DHCP
[ EthDrv] Entering State: Wait for IP Address
[344023][E][vfs_api.cpp:104] open(): /littlefs/www/index.html does not exist, no permits for creation
[346850][E][vfs_api.cpp:104] open(): /littlefs/www/favicon.ico.gz does not exist, no permits for creation
[WiFiDrv] WiFi Lost the connection to the AP
[WiFiDrv] WiFi Entering State: Connection Failed
[ EthDrv] Entering State: Connecting
[ EthDrv] Connecting to Ethernet using DHCP
[ EthDrv] Entering State: Wait for IP Address
[ EthDrv] Entering State: Connecting
[ EthDrv] Connecting to Ethernet using DHCP
[ EthDrv] Entering State: Wait for IP Address
[ EthDrv] Entering State: Got IP
[   FPPD] Listening on port 32320
[ EthDrv] Ethernet Connected with IP: 192.168.0.41
[645552][E][vfs_api.cpp:104] open(): /littlefs/www/index.html does not exist, no permits for creation
[645901][E][vfs_api.cpp:104] open(): /littlefs/www/esps.css does not exist, no permits for creation
[645989][E][vfs_api.cpp:104] open(): /littlefs/www/esps.js does not exist, no permits for creation
[ WebMgr] WS client connect - 1
[646289][E][vfs_api.cpp:104] open(): /littlefs/www/favicon.ico.gz does not exist, no permits for creation
[FileMgr] Configuration File: '/config.json' saved 599 bytes.
[FileMgr] Configuration File: '/config.json' loaded.
[FileMgr] Configuration File: '/config.json' saved 575 bytes.
[ EthDrv] Entering State: Connecting
[ WebMgr] WS client disconnect - 1
[742527][E][ETH.cpp:425] config(): STA IP could not be configured! Error: 20487
[ EthDrv] Connecting to Ethernet with Static IP
[ EthDrv] Entering State: Wait for IP Address
[ EthDrv] Entering State: Got IP
[   FPPD] Listening on port 32320
[ EthDrv] Ethernet Connected with IP: 192.168.0.41
[ WebMgr] WS client connect - 2
[ EthDrv] Entering State: Connecting
[ WebMgr] WS client disconnect - 2
[832527][E][ETH.cpp:425] config(): STA IP could not be configured! Error: 20487
[ EthDrv] Connecting to Ethernet with Static IP
[ EthDrv] Entering State: Wait for IP Address

MartinMueller2003, 2022-11-16:
Based on the log, 
WiFi Connected and got an IP
Wired connected and got an IP and stopped the wifi (expected)
You used a browser connected to the wired IP to change the IP address to a Static Value and the IDF failed to set the static IP. It looks like DHCP worked just fine. 

So the bug is: Cannot set a Static IP on the Wired Ethernet Interface.

I will try to reproduce this on one of my setups. In the meantime, it looks like DHCP is working. If you need a fixed IP for the device you should be able to create a DHCP lease on your router until this gets fixed.

sonofice80, 2022-11-16:
Ok. But when i leave the WiFi sone and go upstairs to my kids room, where i only have wired connection and no WiFi signal i cant log into the User interface.  And it just saus wait for ip adresse in serial monitor.

Tnx for all your effort and time :)

sonofice80, 2022-11-16:
Sorry faulty network cabel on my end. So as you said: « So the bug is: Cannot set a Static IP on the Wired Ethernet Interface»

again tnx for your time and effort

MartinMueller2003, 2022-12-07:
In testing, I am seeing issues with static IP on WiFi interface as well.

MartinMueller2003, 2022-12-09:
I have just verified that it is possible to set the IP address to Static from the UI for both WiFi and Wired connections. There is a bug in the flashtool that does not set the DHCP flag properly. I did notice that the reporting of a failure when setting a static IP is not very good. All you get is a rejected request when the addresses are not set up properly. I initially set the wrong gateway subnet address and the OS rejected my request without explanation. 

At this point, I cannot repeat the issue.

MartinMueller2003, 2023-01-13:
Have you tried the latest build? I would like to close this issue if the latest libs solve your problem.

closed 2023-02-27 by MartinMueller2003