forkineye/ESPixelStick #588: dig-octa does not appear for flash in ESPSFlashTool.jar drop down - build ESPixelStick_Firmware-4.0-ci3315755181

mstang1988, 2022-11-22:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
ESPixelStick_Firmware-4.0-ci3315755181

**Hardware Version**
dig-octa

**Binary release or compiled yourself?**
No

**Operating System (and version)**
OSX Monterey 12.6

**Web Browser (and version)**
Chrome

**Access Point**
Unifi AC-Pro

**Describe the bug**
dig-octa does not appear for flash in ESPSFlashTool.jar drop down to flash so I can only flash the dig quad version of the firmware without walking through the install via command line.

MartinMueller2003, 2022-11-22:
Are you sure you have the latest version of the flash tool? It is present on the latest.

[image]

mstang1988, 2022-11-22:
I used the version that is embedded with the latest CI build available (version published).  I just hacked the json to add it myself and uploaded the firmware.  Let me see if I can download the latest async to the build and if that works.

mstang1988, 2022-11-22:
It looks like main utilizes a version that is 14 months old and was last updated for dig-quad.

https://github.com/forkineye/ESPixelStick/tree/main/dist

https://github.com/forkineye/ESPixelStick/blob/main/dist/ESPSFlashTool.jar

And the definition file as well:
https://github.com/forkineye/ESPixelStick/tree/main/dist/firmware
https://github.com/forkineye/ESPixelStick/blob/main/dist/firmware/firmware.json

MartinMueller2003, 2022-11-22:
Use the version found here. It is always up-to-date:
https://drive.google.com/file/d/1DG3wz8lbYNc371OIbeOqkzMaY5Zd2SKA/view?usp=sharing

MartinMueller2003, 2022-11-30:
Can this issue be closed?

closed 2022-12-05 by MartinMueller2003