forkineye/ESPixelStick #593: Relay wemos fail if over 50 pixels

roboengr2, 2022-11-27:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
4 beta 4

**Hardware Version**
wemos mini

**Binary release or compiled yourself?**
included in jar

**Operating System (and version)**
win 10 21h1

**Web Browser (and version)**
chrome latest nov 2022

**Access Point**
MicroStation m1

**Describe the bug**
When the pixels ws2811 go over 50 the relay out stops working, causes WD issues, and re-sets randomly. I change to go back to 50 or less and it works fine on vixen.   it seems the relay routines are taking too long to execute.  please evaluate for the ISR running too many routines.  seems the FPP is listening and there seems to be no way to shut it off.  need ways to reduce code to accommodate Relay and ws2812 string.  I want to run 136 pixels on the wemos but can't.  Please explain or advise a way to reduce the code being executed.

MartinMueller2003, 2022-11-27:
Where did you get the image to load into the ESP? The V4 ESP does not use an ISR for the relay outputs.

roboengr2, 2022-11-27:
From my phone:

Hello, nice to hear from you so quick.  First, thank you for efforts on
such a nice project along with Shelby and others.  I was a beginning member
of doityourselfchristmas.  I orinally started with p shorts with the renard
64 pic chips.  Then kromby arduino, then the espixelstick using the gece
strings from dark... hack.  There is more but I will spare you of my
involvement but in 2008, 14 years ago, I suggested the original plan in a
drawing combining others work as a master plan for the pixel controls. I am
humble but like the technology ride.  I did not push but what an
accomplishment of a lot of people in the community.  The falcon player is a
big deal too which was a vision then.  We are blessed and you have done a
lot as well.  I am a computer software, Robotics, industrial
automation(before IIOT and IOT, MQTT now), and electronic engineer as a
trade, btw. With a masters degree, this i tell you so that you know more
about my background.

So for your question.  I looked through the code quickly.  I am for sure
less knowledgeable then you.  So I must be mistaken on my first assesment
of the ISR use.  It was assumed and appologize for that. I have run into
the limits of the WD and coding on the esp family in various projects for
hire in the recent past too.  Let's start over on the issue I experienced,
shall we.

I have tried to learn how to implement the espixelstick code for using the
relay.  My intention is to use 1 relay to do the work I need;for now.  I
have found little information on sending the bit from vixen 3 the latest
version and the espixelstick v4 beta4.  I am using a set level element in
vixen to toggle the channel between 0 and 255.  The device hardware I am
using is a wemos d1 mini an esp8266-12f I believe.  I have it loaded with
the espixelsick v4 beta 4 version using the java jar uploader.  In which
after experimentation, learning on my own, to set it up. Very scarce
documentation found.    Btw. I am using John chur...(spelling escapes me)
setup of the obbiqu... nanostation Loco m2 set up as an access point.
Using it for about 3 years now with good results.  Vixen 3 was running at
25ms refresh rate. Perhaps this is agresive but provided a good show.

I have been runing espixelstick on v3.1 on about 10 units.  3.2 gave me
heart burn whhen Shellby M released it before the forks.  Now on the
espixelstick v4 beta4 , I first set the output 2  to relay.  The primary is
ws2811.  When I had the ws2811 set to 1 channel I got the d1 board pin or
gpio5 to toggle.  When I increased the channels to 40 it still worked.
Going above it failed.  Returning it below 40 worked again.  I found a
controlled repeating pattern.  A refresh rate on espixelstick tab "device
setup" worked if around 2ms.  I need 136 pixels or 408 channels with
relay.  The 3.2 espixelstick works.  I think the v4 will work without the
relay also. But Un proven.

So, now when observing the com port of the wemos mini through the java jas
uploader, it showed a watchdog timer reset a few times.  I am figuring
something is causing the code path to be too long for the timeout. I had a
dump I am not sure if that would be something you can use.  I would have to
recreate it again to send the dump/info that was sent.  But if I do
recreate the issue I want to test it with your input.

Thus, so i wish for your input.  Can you instruct me in the proper setup,
if I am in error.  Then I can get you a good test.  I do gave a logic
analyzer a sealce one IF it may help.  I am new to mqtt so I would need to
learn more to use it on this device.  I have used topics on a pub-sub
protocol other places.

I have hoped the esp8266 could handle the load.  But if I need the esp32
dual core then advise me on that too.  I am concerned about what esp32
hardware to purchase on a hobby project budget.  The lonlin esp32 pro is up
there. Is a wemos esp32 mini from Amazon a recommendation? Please advise.

Finally, Please advise me on what to do to help you help me by providing a
test plan.  For repayment I could help document the relay setup. If not
already done, i can provde an example for the community to use.

Let me know your thoughts.

Roboengr

MartinMueller2003, 2022-11-27:
Good Day. First thing to note is that the beta 4 version is very old and has a LOT of issues. Please use an image from this link: 
https://drive.google.com/file/d/1DG3wz8lbYNc371OIbeOqkzMaY5Zd2SKA/view?usp=sharing

As for channel order, V4 leaves you no choices here. Your pixels will go first (On an 8266 that would be less than 1000), followed by your relay channels. You will need to set the GPIO values for the channels via the advanced mode on the UI.

Pixels DO use ISRs to output the data and the number of pixels supported is controlled by available ram, which for the 8266 is pretty small. 

Relay channels use a polling loop and have appropriate yield points to prevent the WDT from getting triggered. There are two kinds of Relay channels available. The built in PWM outputs and 16 additional relay / Servo outputs via an I2C connected PCA9685 module. Many features are not available until you go to an ESP32 based card. If you get an ESP32 module WITH PSRAM the number of supported pixels climbs to well over 5000 and the number of pixel ports goes to 8 (See the Dig-Octa from QuinLed).

roboengr2, 2022-11-27:
Martin,

Thanks again for getting back quickly.  I am trying to work with you as
quickly as I can while hoping you don't get busy with other things.

I download the link and tried it. not a good result  I have attached what I
am getting when uploaded to the Wemaos DI mini with the
8266-12f(esp8266mod). it's repeating and seems to be something to do with
booting.

I have tried a lot of things but get back to this attached outcome.  I sort
of like the new flash tool, not sure I am using it correctly.  I have found
that the use of DHCP has inverted as well as the other checkboxes.  I know
they are because the DCHP unchecked does use DHCP as reported in the
startup serial debug data. In the attached file I used my Static
addressing.

the exception 28 is something I can not figure out and was hoping you know.

This dev you sent to download is still intended to work on the 8266
correct?  I noted there are more selections on hardware in the new tool.  I
don't have a spare esp32 yet to try. but thank you for your suggestion on
the site your provided.  the  ESP32 WROOM32 module version “E” with
ESP32-D0WD(-V3)
chip seems to be the one to use, am I correct?  but you have the d32 mini's
too.  I am still trying to keep the foot print to the mini to fit a 1.5
inch PVC pipe enclosure.

Can you get me past the repeated reboot? any suggestions?

Thanks,

ROBOENGR

MartinMueller2003, 2022-11-27:
You did not attach the output from the flash tool. Just click the save logs button and send me the file.

roboengr2, 2022-11-27:
attachment 2 send

MartinMueller2003, 2022-11-28:
I am still not seeing any attachments. Are you zipping the file before you attach it? You may be hitting a size limit.

roboengr2, 2022-11-29:
Something is striping the attachment. I think it is because it does not
have a .txt on it and was blank.  so that is something the tool needs to do
add a .txt to the file type if this works.

System Output: 

 bin/win32/mklittlefs -c fs/ -p 256 -b 8192 -s 2072576 firmware/filesystem.bin
-= Building Filesystem Image =-
 bin/win32/mklittlefs -c fs/ -p 256 -b 8192 -s 2072576 firmware/filesystem.bin
/config.json
/UpdRecipe.json
/www/alexa.html.gz
/www/apa102.html.gz
/www/artnet.html.gz
/www/ddp.html.gz
/www/dmx.html.gz
/www/e1_31.html.gz
/www/effects.html.gz
/www/esps.css.gz
/www/esps.js.gz
/www/favicon.ico
/www/fpp_remote.html.gz
/www/gece.html.gz
/www/gs8208.html.gz
/www/index.html.gz
/www/mqtt.html.gz
/www/relay.html.gz
/www/renard.html.gz
/www/serial.html.gz
/www/servo_pca9685.html.gz
/www/tls3001.html.gz
/www/tm1814.html.gz
/www/ucs1903.html.gz
/www/ucs8903.html.gz
/www/ws2801.html.gz
/www/ws2811.html.gz
 bin/win32/python3/python bin/upload.py --chip esp8266 --baud 460800 --port COM7 erase_flash
/config.json
/UpdRecipe.json
/www/alexa.html.gz
/www/apa102.html.gz
/www/artnet.html.gz
/www/ddp.html.gz
/www/dmx.html.gz
/www/e1_31.html.gz
/www/effects.html.gz
/www/esps.css.gz
/www/esps.js.gz
/www/favicon.ico
/www/fpp_remote.html.gz
/www/gece.html.gz
/www/gs8208.html.gz
/www/index.html.gz
/www/mqtt.html.gz
/www/relay.html.gz
/www/renard.html.gz
/www/serial.html.gz
/www/servo_pca9685.html.gz
/www/tls3001.html.gz
/www/tm1814.html.gz
/www/ucs1903.html.gz
/www/ucs8903.html.gz
/www/ws2801.html.gz
/www/ws2811.html.gz
Done

-= Erasing ESP Flash =-
 bin/win32/python3/python bin/upload.py --chip esp8266 --baud 460800 --port COM7 erase_flash
esptool.py v2.8
Serial port COM7
[... 558 lines cut ...]
11:51:40: <<<stack<<<
11:51:40: 
11:51:40: --------------- CUT HERE FOR EXCEPTION DECODER ---------------
11:51:40: 
11:51:40:  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
11:51:40: 
11:51:40: load 0x4010f000, len 3460, room 16 
11:51:40: tail 4
11:51:40: chksum 0xcc
11:51:40: load 0x3fff20b8, len 40, room 4 
11:51:40: tail 4
11:51:40: chksum 0xc9
11:51:40: csum 0xc9
11:51:40: v00088fb0
11:51:40: ~ld
11:51:40: 
11:51:40: [    ESP] ESPixelStick v4.x-dev (Nov 18 2022 - 12:35:28)
11:51:40: [    ESP] SDK:2.2.2-dev(38a443e)/Core:3.0.2=30002000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:6105635
11:51:40: [FileMgr] Flash file system initialized.
11:51:40: [FileMgr] No SD card installed
11:51:40: [FileMgr] Configuration File: '/config.json' loaded.
11:51:40: [FileMgr] Configuration File: '/input_config.json' loaded.
11:51:40: [FileMgr] Configuration File: '/output_config.json' loaded.
11:51:40: [FileMgr] Configuration File: '/config.json' saved 377 bytes.
11:51:41:

roboengr2, 2022-11-29:
if it does not make it is there another suggestion for getting you the
attachment?  I am zipping it up for this email. tell me which one makes it
for future reference.

roboengr2, 2022-11-30:
> Log made it. As an experiment, turn off static IP. Use DHCP and see if it
> is stable.

DHCP with out of the box settings from the file you send for me to use.
Does about the same.

System Output: 

 bin/win32/mklittlefs -c fs/ -p 256 -b 8192 -s 2072576 firmware/filesystem.bin
-= Building Filesystem Image =-
 bin/win32/mklittlefs -c fs/ -p 256 -b 8192 -s 2072576 firmware/filesystem.bin
/config.json
/UpdRecipe.json
/www/alexa.html.gz
/www/apa102.html.gz
/www/artnet.html.gz
/www/ddp.html.gz
/www/dmx.html.gz
/www/e1_31.html.gz
/www/effects.html.gz
/www/esps.css.gz
/www/esps.js.gz
/www/favicon.ico
/www/fpp_remote.html.gz
/www/gece.html.gz
/www/gs8208.html.gz
/www/index.html.gz
/www/mqtt.html.gz
/www/relay.html.gz
/www/renard.html.gz
/www/serial.html.gz
/www/servo_pca9685.html.gz
/www/tls3001.html.gz
/www/tm1814.html.gz
/www/ucs1903.html.gz
/www/ucs8903.html.gz
/www/ws2801.html.gz
/www/ws2811.html.gz
/config.json
/UpdRecipe.json
/www/alexa.html.gz
/www/apa102.html.gz
/www/artnet.html.gz
/www/ddp.html.gz
/www/dmx.html.gz
/www/e1_31.html.gz
/www/effects.html.gz
/www/esps.css.gz
/www/esps.js.gz
/www/favicon.ico
/www/fpp_remote.html.gz
/www/gece.html.gz
/www/gs8208.html.gz
/www/index.html.gz
/www/mqtt.html.gz
/www/relay.html.gz
/www/renard.html.gz
/www/serial.html.gz
/www/servo_pca9685.html.gz
/www/tls3001.html.gz
/www/tm1814.html.gz
/www/ucs1903.html.gz
/www/ucs8903.html.gz
/www/ws2801.html.gz
/www/ws2811.html.gz
 bin/win32/python3/python bin/upload.py --chip esp8266 --baud 460800 --port COM7 erase_flash
Done

-= Erasing ESP Flash =-
 bin/win32/python3/python bin/upload.py --chip esp8266 --baud 460800 --port COM7 erase_flash
esptool.py v2.8
Serial port COM7
esptool.py v2.8
[... 1277 lines cut ...]
19:50:19: <<<stack<<<
19:50:19: 
19:50:19: --------------- CUT HERE FOR EXCEPTION DECODER ---------------
19:50:19: 
19:50:19:  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
19:50:19: 
19:50:19: load 0x4010f000, len 3460, room 16 
19:50:19: tail 4
19:50:19: chksum 0xcc
19:50:19: load 0x3fff20b8, len 40, room 4 
19:50:19: tail 4
19:50:19: chksum 0xc9
19:50:19: csum 0xc9
19:50:19: v00088fb0
19:50:19: ~ld
19:50:19: 
19:50:20: [    ESP] ESPixelStick v4.x-dev (Nov 18 2022 - 12:35:28)
19:50:20: [    ESP] SDK:2.2.2-dev(38a443e)/Core:3.0.2=30002000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:6105635
19:50:20: [FileMgr] Flash file system initialized.
19:50:20: [FileMgr] No SD card installed
19:50:20: [FileMgr] Configuration File: '/config.json' loaded.
19:50:20: [FileMgr] Configuration File: '/input_config.json' loaded.
19:50:20: [FileMgr] Configuration File: '/output_config.json' loaded.
19:50:20: [FileMgr] Configuration File: '/config.json' saved 370 bytes.
19:50:20:

MartinMueller2003, 2022-11-30:
Interesting. The flash tool has a download config option would you please click that. After it finishes you will have a download directory with a few .json file in it. Please attach those files to this conversation.

roboengr2, 2022-12-01:
Martin,

this is what I am getting below.  the download directory is created but
empty.  I tried a number of times with the DHCP option

Downloading Configuration Files from the ESP
 bin/win32/python3/python bin/upload.py --chip esp8266 --baud 460800 --port
COM7 --before default_reset --after hard_reset read_flash 0x200000 2072576
downloaded/Wemos_D1_Mini_0x200000_2072576.bin
-= Retreiving Filesystem Image =-
 bin/win32/python3/python bin/upload.py --chip esp8266 --baud 460800 --port
COM7 --before default_reset --after hard_reset read_flash 0x200000 2072576
downloaded/Wemos_D1_Mini_0x200000_2072576.bin
esptool.py v2.8
Serial port COM7
esptool.py v2.8
Serial port COM7
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: ac:0b:fb:dc:fc:c2
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Traceback (most recent call last):
  File "E:\projects\espixelstick 2020\dist\dist\bin\upload.py", line 19, in
<module>
    esptool.main(sys.argv)
  File "E:/projects/espixelstick 2020/dist/dist/bin/esptool\esptool.py",
line 2938, in main
    operation_func(esp, args)
  File "E:/projects/espixelstick 2020/dist/dist/bin/esptool\esptool.py",
line 2551, in read_flash
    data = esp.read_flash(args.address, args.size, flash_progress)
  File "E:/projects/espixelstick 2020/dist/dist/bin/esptool\esptool.py",
line 104, in inner
    return func(*args, **kwargs)
  File "E:/projects/espixelstick 2020/dist/dist/bin/esptool\esptool.py",
line 741, in read_flash
    raise FatalError('Corrupt data, expected 0x%x bytes but received 0x%x
bytes' % (self.FLASH_SECTOR_SIZE, len(p)))
esptool.FatalError: Corrupt data, expected 0x1000 bytes but received 0xc01
bytes
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: ac:0b:fb:dc:fc:c2
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Traceback (most recent call last):
  File "E:\projects\espixelstick 2020\dist\dist\bin\upload.py", line 19, in
<module>
    esptool.main(sys.argv)
  File "E:/projects/espixelstick 2020/dist/dist/bin/esptool\esptool.py",
line 2938, in main
    operation_func(esp, args)
  File "E:/projects/espixelstick 2020/dist/dist/bin/esptool\esptool.py",
line 2551, in read_flash
    data = esp.read_flash(args.address, args.size, flash_progress)
  File "E:/projects/espixelstick 2020/dist/dist/bin/esptool\esptool.py",
line 104, in inner
    return func(*args, **kwargs)
  File "E:/projects/espixelstick 2020/dist/dist/bin/esptool\esptool.py",
[... 361 lines cut ...]
21:21:58: <<<stack<<<
21:21:58: 
21:21:58: --------------- CUT HERE FOR EXCEPTION DECODER ---------------
21:21:58: 
21:21:58:  ets Jan  8 2013,rst cause:2, boot mode:(3,7)
21:21:58: 
21:21:58: load 0x4010f000, len 3460, room 16 
21:21:58: tail 4
21:21:58: chksum 0xcc
21:21:58: load 0x3fff20b8, len 40, room 4 
21:21:58: tail 4
21:21:58: chksum 0xc9
21:21:58: csum 0xc9
21:21:58: v00088fb0
21:21:58: ~ld
21:21:58: 
21:21:58: [    ESP] ESPixelStick v4.x-dev (Nov 18 2022 - 12:35:28)
21:21:58: [    ESP] SDK:2.2.2-dev(38a443e)/Core:3.0.2=30002000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:6105635
21:21:58: [FileMgr] Flash file system initialized.
21:21:58: [FileMgr] No SD card installed
21:21:58: [FileMgr] Configuration File: '/config.json' loaded.
21:21:58: [FileMgr] Configuration File: '/input_config.json' loaded.
21:21:58: [FileMgr] Configuration File: '/output_config.json' loaded.
21:21:58: [FileMgr] Configuration File: '/config.json' saved 370 bytes.
21:21:58:

roboengr2, 2022-12-01:
these popups too.

 dist after test.zip
<https://drive.google.com/file/d/10-IzJwZs6cHrynvPhoF5nKwZPT22enBD/view?usp=drive_web>

roboengr2, 2022-12-01:
I go back to the espixelsitck v4 beta4 and this is the status and serial
lists attached

then go to the dist you send and read the download

and still will not download.  I hope this may contain clues

-= Building Filesystem Image =-
 bin/win32/mklittlefs -c fs/ -p 256 -b 8192 -s 2072576 firmware/filesystem.bin
/config.json
/UpdRecipe.json
/www/alexa.html.gz
/www/apa102.html.gz
/www/artnet.html.gz
/www/ddp.html.gz
/www/dmx.html.gz
/www/e1_31.html.gz
/www/effects.html.gz
/www/esps.css.gz
/www/esps.js.gz
/www/favicon.ico
/www/fpp_remote.html.gz
/www/gece.html.gz
/www/index.html.gz
/www/mqtt.html.gz
/www/relay.html.gz
/www/renard.html.gz
/www/serial.html.gz
/www/servo_pca9685.html.gz
/www/tls3001.html.gz
/www/tm1814.html.gz
/www/ws2801.html.gz
/www/ws2811.html.gz
Done

-= Programming ESP =-
 bin/win32/python3/python bin/upload.py --chip esp8266 --baud 460800 --port COM7 erase_flash
esptool.py v2.8
Serial port COM7
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: ac:0b:fb:dc:fc:c2
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Erasing flash (this may take a while)...
Chip erase completed successfully in 7.2s
Hard resetting via RTS pin...
Done
 bin/win32/python3/python bin/upload.py --chip esp8266 --baud 460800 --port COM7 --before default_reset --after hard_reset write_flash 0x0 firmware/esp8266/d1_mini-app.bin 0x200000 firmware/filesystem.bin
esptool.py v2.8
Serial port COM7
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: ac:0b:fb:dc:fc:c2
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 561696 bytes to 386094...

[... 157 lines cut ...]
21:33:16: [ WebMgr] WS client disconnect - 48
21:33:18: [ WebMgr] WS client connect - 49
21:33:18: ????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????rl l?r$?n?l?b|???rb?rl l?r$?n?l?b|???rb?rl l?r$?n?l?b|???rb?rl l?r$?n?l?b|???rb?b?nnlnnbbp?$blrlp?n??l?bn?n??b??nn'l?l`?nn$`nr???nrr?`p?n?r?bbn?nb??nn'l`?nn$`nr???nrl`r??nrl`?l??l`??n?`
21:33:36: [    ESP] ESPixelStick v4.0-beta4 (Nov 13 2021 - 14:53:28)
21:33:36: [    ESP] SDK:2.2.2-dev(38a443e)/Core:3.0.2=30002000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:6105635
21:33:36: [FileMgr] Flash file system initialized.
21:33:36: [FileMgr] No SD card installed
21:33:36: [FileMgr] Configuration File: '/config.json' loaded.
21:33:36: [FileMgr] Configuration File: '/output_config.json' loaded.
21:33:36: [FileMgr] Configuration File: '/input_config.json' loaded.
21:33:36: [WiFiMgr] WiFi Entering State: Connecting Using Config Credentials
21:33:37: [WiFiMgr] Using Static IP
21:33:37: [WiFiMgr] Connecting to 'RedNose' as test
21:33:37: [WiFiMgr] WiFi Entering State: Connected To AP
21:33:45: [WiFiMgr] Connected with IP: 192.168.25.91
21:33:45: [ WebMgr] Web server listening on port 80
21:33:45: [   FPPD] Listening on port 32320
21:33:45: [ WebMgr] WS client connect - 1
21:33:48: [ WebMgr] WS client connect - 2
21:33:53: [ WebMgr] WS client disconnect - 1
21:33:54: [ WebMgr] WS client disconnect - 2
21:33:58: [ WebMgr] WS client connect - 3
21:34:00: [ WebMgr] WS client disconnect - 3
21:34:04: [ WebMgr] WS client connect - 4
21:34:04:

roboengr2, 2022-12-01:
I don't know if this is anything but the mini firmware are different
size files. Do you think this is a clue? the dist seems to be suspect on
the size difference.

roboengr2, 2022-12-04:
any solutions to this or suggestions?

MartinMueller2003, 2022-12-09:
A fix for the crash I was seeing is now in my dist.zip along with a new effect called transitions that allows you to create a list of colors that the ESP morphs through over time.

roboengr2, 2023-02-12:
> I just released a new version that adds a fixe for the Relay Outputs to
> set them properly. The same link you have been using will give you the new
> version

Thanks!

I still can't get relays to work on the esp8266 wemos d1 mini.  At this
point it maybe me.  What values do I use for the GPIO?  I tried the say 16
for d0. I have has it crash on some settings.  Is there documentation on
installing from the release files to config of static ip to buttons to
press on the Downloader jar file.  To setting up relay out put for xlights
and vixen and maybe other uses even with the ffp SD card capability.

Thanks for your help

MartinMueller2003, 2023-02-13:
There are no videos available. As for valid GPIOs, GPIO16 should work fine. Not sure why it is crashing for you. I will load it up and give it a try.

MartinMueller2003, 2023-02-27:
FYI: This weekend I tested with 800 pixels and the relay ports enabled. There were some issues caused by the unused relay port definitions. I had to change the default GPIO to -1 so that no other GPIOs would be accidentally triggered. The old setting was GPIO zero and using that seemed to have some conflicts.

closed 2023-02-27 by MartinMueller2003