forkineye/ESPixelStick #603: ESPixelStick 4.0-beta4 no file management tab

Billybojangles01, 2022-12-10:
--------- Instructions --------
Please provide answers directly beloESPixelStickw each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
4.0-beta4

**Hardware Version**
Dig Octa 

**Binary release or compiled yourself?**
Binary release

**Operating System (and version)**
Windows 10

**Web Browser (and version)**
Any

**Access Point**

**Describe the bug** Trying to send xlights seq to device but since there is no file management tab I can only play remote seq.
SD Card is formated with exFat

MartinMueller2003, 2022-12-10:
I just looked at the config and values up to 40 are allowed for MISO. I set it to 35 and then back to 46 and everything on the octa is working as expected. If you are using the octa image from the installer, you should not need to change the GPIO settings.

[image]

Billybojangles01, 2022-12-10:
How can I check my config?  It will not accept anything over 24 for me.

MartinMueller2003, 2022-12-10:
First, grab the latest dist.zip from my googledrive. That is the exact version I tested. Then post the admin page like you did. I suspect there is something cached on your UI.

Billybojangles01, 2022-12-11:
> First, grab the latest dist.zip from my googledrive. That is the exact version I tested. Then post the admin page like you did. I suspect there is something cached on your UI.

I can do that.. Where do I find the link to your GoogleDrive?

MartinMueller2003, 2022-12-11:
https://drive.google.com/file/d/1DG3wz8lbYNc371OIbeOqkzMaY5Zd2SKA/view?usp=sharing

closed 2023-02-27 by MartinMueller2003