forkineye/ESPixelStick #606: No light output from Xlights sequence via FPP Remote

Sab66, 2022-12-20:
Was working until yesterday.  Edited Sequence in Xlights and uploaded it via FPP Connect to 7 ESPixelsticks running FW Version 4.0-beta4.  One of the controllers outputs correctly but the other 6 do not.  Uploaded original sequence but no change.  i can run the Effects from each controller and it outputs to the lights.  See packets increasing on  each controllers Statius page as well as the correct FPP Remote status sequence and timing.  Can produce screenshots if necessary.  Checked settings in Xlights, FPP and on each controller to make sure that everything matched.  Only edited a sequence and once uploaded, lights failed to come on with the exception of one controller

MartinMueller2003, 2022-12-20:
This really sounds like an xlights issue. If you use sACNview to create a data stream, do the ESPs do the right thing?

Sab66, 2022-12-20:
Never used it before but will try to figure it out and report back.  If it is an Xlights problem, sorry in advance for wasting your time

MartinMueller2003, 2022-12-20:
So it uploaded fseq files and you are in remote FPP mode?

MartinMueller2003, 2022-12-20:
Please put the device into advanced mode and then look at the status page to see if there is an error message posted.

Sab66, 2022-12-20:
Yes sequences uploaded to FPP and all 7 controllers.  All controllers set to FPP remote as secondary input and set to Play remote sequence under FSEQ File to Play

MartinMueller2003, 2022-12-20:
Need to see a screenshot of the status screen in admin mode

MartinMueller2003, 2022-12-20:
advanced mode

Sab66, 2022-12-20:
[image]

Sab66, 2022-12-20:
[image]

MartinMueller2003, 2022-12-21:
You did not turn on advanced mode prior to taking the screenshots. I need a special message that gets displayed when advanced mode is turned on

MartinMueller2003, 2022-12-21:
and the time remaining is a dead giveaway that the fseq is corrupted

Sab66, 2022-12-21:
I did click Advanced mode in the Admin screen, but saw nowhere to save it.  I tried a brand new sequence with similar results.  Not sure what to try to be honest
Thanks for your time.

MartinMueller2003, 2022-12-21:
advanced mode cannot be saved. that is on purpose. click on advanced mode, start your sequence, click on home/status page. an error message will be displayed.

MartinMueller2003, 2022-12-21:
Also, please send me a copy of that file. I would like to run it through some tools we have that can tell us what is wrong with the file

adamcoulombe, 2023-01-06:
I may be having the same, or similar issue.
Architecture ESP32, build 4.0-ci3659911785. FSEQ rendered by latest xlights.

Any fseq I upload (either fpp connect or manually through espixelstick UI), I see the following error
```
ParseFseqFile:: Could not start: pixelsticktest.fseq File does not contain enough data to meet the Stated Channel Count * Number of Frames value. Expected 3060000, Got: 816128
```
fseq is attached
[pixelsticktest.zip](https://github.com/forkineye/ESPixelStick/files/10362772/pixelsticktest.zip)

MartinMueller2003, 2023-01-06:
What is the full fseq? You want an fseq that has only the data for this esp

adamcoulombe, 2023-01-07:
> What is the full fseq? You want an fseq that has only the data for this esp

Right, I was just trying different things to see what might work. By “full fseq” I was referring to the non-sparse, uncompressed file that xlights generates when I save. If that isn’t helpful information though you can just ignore it :)

adamcoulombe, 2023-01-07:
Fair enough; the error referenced above though occurs when playing the attached fseq which was uploaded through fpp connect and IS the sparse version

MartinMueller2003, 2023-01-07:
Yes and it is corrupted. xLights is able to save the sparse version to disk and you should be able to transfer it to the ESP. That should allow you to make progress.

adamcoulombe, 2023-01-08:
I am having _some_ luck getting fseq's to light up if I keep my total pixel count on the ESP32 below 255. If I configure above that, It is almost guaranteed to create the `ParseFseqFile` error mentioned above.

You mentioned xlights can save the sparse file it sends to the ESP to my hard drive. Where do I find that?

I see you can change a setting for FSEWQ version under preferences. There is an option for sparse but I tried that one but the ESP32 complained it wasn't uncompressed in the serial monitor.
[image]

MartinMueller2003, 2023-01-08:
you will need to ask on the xlights zoom room. i am a vixen user and have limited knowledge about xlights

adamcoulombe, 2023-01-08:
OK, your suggestion to export the sparse file led me to a fruitful comparison. I couldnt find a way to directly export the sparse/uncompressed file onto my hard drive, but I set up a fresh raspberry pi/fpp with identical configuration (start channels, outputs, and number of pixels) and uploaded the sparse/uncompressed file from xlights to the pi. 

I then copied that exact fseq file from the Pi onto the esp32.

The result was that the esp32 gives the `ParseFseqFile` error while the Raspberry Pi plays it fine.

A copy of that fseq file is attached.
[pixelsticktest(2).zip](https://github.com/forkineye/ESPixelStick/files/10367605/pixelsticktest.2.zip)

MartinMueller2003, 2023-01-09:
Now that is something I can look into. It indicates some form of truncation in the transfer process.

MartinMueller2003, 2023-01-09:
I just transferred the attached file via my chrome browser (multiple times) with no issues. The file plays fine. I used the ESP32 Quinled AE+ as the platform.

adamcoulombe, 2023-01-10:
> I just transferred the attached file via my chrome browser (multiple times) with no issues. The file plays fine. I used the ESP32 Quinled AE+ as the platform.

Martin, I really appreciate you taking the time to test that for me.  Based on this, I suspect what I am actually experiencing is a hardware issue causing corruption of the fseq file as it saves the file to the SD card.

MartinMueller2003, 2023-01-11:
I tested sending the attached file via fpp connect multiple times yesterday and never had it truncated. I have made a few changes to the download code in an attempt to make the transfers go faster, but the changes had no real impact. I did make sure I updated xLights to the latest version, but that should not have made any difference. FYI: This means that the only remaining difference is how the file is created and handled by xLights.

MartinMueller2003, 2023-01-13:
Are you still seeing the issue using the latest code?

adamcoulombe, 2023-01-20:
> Are you still seeing the issue using the latest code?

I tried on some different hardware and I was able to upload consistently without the file getting corrupted. I think it’s safe to say the issue I was seeing was being caused by faulty hardware

MartinMueller2003, 2023-01-20:
I am not seeing this.

closed 2023-01-20 by MartinMueller2003