forkineye/ESPixelStick #607: DIG-QUAD with AE+ - Is it possible for all 8 outputs?

onewithhammer, 2022-12-21:
**ESPixelStick Firmware Version**
Firmware ESPixelStick: 4.0-ci1789220604
Build: Dec 12, 2022

**Hardware Version**
DIG-QUAD with AE+

**Binary release or compiled yourself?**
Latest DEV release from Martin Mueller share

**Describe the bug**
Downloaded and flashed AE+ with firmware. Selected Hardware “QUINLED QUAD AE+”.

The DIG-QUAD has 4 LED channel + 5 channel (GPIO 15, QR1 on green terminal).

If you add an AE+ that provides 3 more LED channels for a total of 8 outputs.

When I go to configure the device.  The firmware UI provides up to 8 outputs but "Output 8" only has the option of "Disabled" or "Relay".

Is it possible to configure the 8th LED output?  If so, how?

onewithhammer, 2022-12-21:
I currently have this setup and I'm available to provide any information you need.  I'm also available via Discord or zoom call.

MartinMueller2003, 2022-12-21:
You can reach me on discord. I will have to look at the compiled in settings. I am MartinMueller2003

onewithhammer, 2022-12-21:
What is your tag ID?  Here is my Discord: TheChristmasLightGuy#6333

onewithhammer, 2022-12-21:
Nevermind - I found you and sent a request.

MartinMueller2003, 2023-01-09:
It seems that when greater than 6 ports are active there is a flashing on the UART ports.

MartinMueller2003, 2023-01-13:
I have reduced the ISR time by 9% and that has reduced the frequency of flashing.

MartinMueller2003, 2023-02-14:
Still working on this. I have refactored the RMT driver and am now reduced to a single data corruption event.

MartinMueller2003, 2023-02-23:
Did a major restructure of the ESP32 output system and seem to have this working.  Being tested now.

MartinMueller2003, 2023-02-27:
At this time I think the answer to the question in the title is "Yes" :)

MartinMueller2003, 2023-02-27:
Fixes for this are in the latest pull request.

MartinMueller2003, 2023-03-08:
TheChristmasLightGuy — Today at 2:45 PM
I couldn't reproduce either issue
and I tried for several days
I think you can close the ticket.

closed 2023-03-08 by MartinMueller2003