forkineye/ESPixelStick #609: ESPixelstick / Relay Output / Wemos D1 / Seems inconsistent on startup

emccarron, 2023-01-09:
Flashed with 3.2.
7 relays configured
After setup, relays work as expected
Upon reboot, relays do not default to expected state
Occasionally channel locks up and I will need to send DDP on that channel to 255, then back to 0 to reestablish control over that channel
Tried multiple devices
Disabling and re-enabling the relay output often clears issue

Anything to look for?

MartinMueller2003, 2023-01-09:
Please post an image of the config. Alternatively, you can download the config using the flashtool and attach it here.

emccarron, 2023-01-10:
The .json settings backup from the web interface?  Not sure I see how to download from the device using the flash tool?

emccarron, 2023-01-10:
Woops, found it in the dist version of the flash tool... Trying now.

emccarron, 2023-01-10:
[Wemos_D1_Mini_0x200000_2072576.zip](https://github.com/forkineye/ESPixelStick/files/10378603/Wemos_D1_Mini_0x200000_2072576.zip)

All the files it created...  I'm not actually running the dist version on the device... Should I be using that one instead of beta 4?

emccarron, 2023-01-10:
[Wemos_D1_Mini_0x200000_2072576.zip](https://github.com/forkineye/ESPixelStick/files/10378752/Wemos_D1_Mini_0x200000_2072576.zip)

Image attached.

MartinMueller2003, 2023-01-12:
What I need is the .json files. You zipped up the bin file

MartinMueller2003, 2023-01-12:
"After setup, relays work as expected
Upon reboot, relays do not default to expected state"

What is your expected state?

emccarron, 2023-01-13:
[json.zip](https://github.com/forkineye/ESPixelStick/files/10406408/json.zip)

"What is your expected state?"

The state shown in the diagnostics tab.  I have diagnostics open, and it shows only channel 1 on.  Physical relay outputs show 1,3,4,6,7,8 on.  I'm using xlights to send DDP data.

If I manually drive one of the channels  on, then off, it will drive the relay correctly, and the diagnostic channel will sync with the relay.

emccarron, 2023-01-13:
It's almost like it's looking for the signal transition to switch the relay instead of the current state of the DDP output.

emccarron, 2023-01-13:
Quick video.  Power up ESP8266 running ESPixelstick.  Relays come up in random state.  DDP is off.  Turn on DDP.  Diagnostics reflects channel state.

Note channel 2 relay does not respond despite diagnostic status showing change of state.

Note all other 'on' channels do not reflect diagnostic state until channel output is driven from one extreme to another, then it reacts as I'd expect.

Disabling channel 2. then re enabling it seems to bring that channel back.

Turning off DDP drops all relay outputs.

emccarron, 2023-01-13:
Bad video attached. Power up ESP8266 running ESPixelstick. Relays come up in random state. DDP is off. Turn on DDP. Diagnostics reflects channel state.

Note channel 2 relay does not respond despite diagnostic status showing change of state.

Note all other 'on' channels do not reflect diagnostic state until channel output is driven from one extreme to another, then it reacts as I'd expect.

Disabling channel 2. then re enabling it seems to bring that channel back.

Turning off DDP drops all relay outputs.

It seems once they've been cycled, they're fine.  I assume the DDP sender is OK since the diagnostic screen seems normal.

https://user-images.githubusercontent.com/5301465/212233634-bc09c7de-253a-49c5-900d-0f61ec6b109e.mp4

emccarron, 2023-01-13:
The misbehaving channel 2 seems to not be the channel; it follows whichever channel is GPIO 0.

MartinMueller2003, 2023-01-13:
GPIO is the programming pin and needs to be used carefully. You need to make sure it has a pullup on it. That means you want you relay outputs to all have pullups. That means you want your relays to use a "LOW" output to indicate "ON". That will take care of random issues at boot time. I am concerned with random issues at run time but you do not seem to be describing those. I also need to know which GPIOs you are using for which channel. You should NOT be using the GPIOs assigned to the UART 0 TX and RX pins. 

If you send me the output config json file then I can verify how you have configured the system.

MartinMueller2003, 2023-01-13:
I did find a bug that did not always cause the output to be correct. I also added relay status to the status page so you do not need to guess what the output is doing based on the brightness of the diags page.

https://drive.google.com/file/d/1DG3wz8lbYNc371OIbeOqkzMaY5Zd2SKA/view?usp=share_link

MartinMueller2003, 2023-01-13:
`Note channel 2 relay does not respond despite diagnostic status showing change of state.`
- I cannot address this without the configuration. However, if this is GPIO 0 then the previous answer tells you why

`Note all other 'on' channels do not reflect diagnostic state until channel output is driven from one extreme to another, then it reacts as I'd expect.`
- I have fixed this defect and submitted the change to main

`Disabling channel 2. then re enabling it seems to bring that channel back.`
- Again, if GPIO 0 then this is a possible issue.

`Turning off DDP drops all relay outputs.`
- This is per design. All outputs revert to "OFF" after the input data stream terminates.

`It seems once they've been cycled, they're fine. I assume the DDP sender is OK since the diagnostic screen seems normal.`
- This is the same root cause as the first item

emccarron, 2023-01-13:
Nice!  I'll stop using GPIO 0  just in case.  I should have said that turning off DDP causes the relays to drop as expected -- I figured that was normal.

Thanks again!

Ed

emccarron, 2023-01-13:
Had a chance to play.  Downloads, but UI issues.  I suspect my cheap Wemos D1 is the issue.

Will retry with higher quality device.

MartinMueller2003, 2023-02-14:
Any update on your testing?

MartinMueller2003, 2023-03-09:
I think all issues have been addressed and all changes have been integrated in main.

closed 2023-03-09 by MartinMueller2003