forkineye/ESPixelStick #612: Quinled AE+_SD + 3x - no file management with a 32GB SD card

gabi68, 2023-01-20:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
ESPixelStick 4.0-dev - Development Release

**Hardware Version**
QuinLed AE+_SD + 3x

**Binary release or compiled yourself?**
Binary from Google Drive

**Operating System (and version)**
Win 10  64bit

**Web Browser (and version)**
Chrome 109.0.5414.75

**Access Point**

**Describe the bug**
Tried to make a SD card work (save and send xlights sequences), but is not recognized therefore no management file tab . Tried 4GB, 32GB, 64GB all formatted Fat32. All pins for SD card are correct (same pins like the table posted on QuinLed website). All checked with advance mode activated.

MartinMueller2003, 2023-01-20:
Please post an image of the Admin page. It has some important info on it.

gabi68, 2023-01-20:
Here you go
[image]

gabi68, 2023-01-20:
And pins
[image]

MartinMueller2003, 2023-01-20:
I have that build on my Quinled AE Plus card and yes I have the same GPIOs and I see the file management screen just fine. Please make sure everything is properly plugged in. It is very easy to mess up the alignment of the pins on the AE Plus extension card.

gabi68, 2023-01-21:
In other words I have a board which is not working as is should. I will have another look on connections which are good since the rainbow pattern is working...

MartinMueller2003, 2023-01-21:
It is possible that the HW is bad. I myself have plugged the AE card in wrong a number of times, so I know it can happen.

MartinMueller2003, 2023-03-10:
Any update? I am not seeing this on any other boards.

MartinMueller2003, 2023-04-13:
No changes made. I suspect a bad or not compatible SD card.

closed 2023-04-13 by MartinMueller2003