forkineye/ESPixelStick #613: ESP32 CAM, crashed when enable output 4 with WS2811 (boot loop)

pakeonoahu, 2023-01-28:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**

ESPixelStick v4.x-dev (Jan 13 2023 - 10:42:28)

**Hardware Version**
ESP32 CAM

**Binary release or compiled yourself?**
Binary dev release.

**Operating System (and version)**

**Web Browser (and version)**
Chrome

**Access Point**

**Describe the bug**
22:49:15: Rebooting...
22:49:15: ets Jun  8 2016 00:22:57
22:49:15: 
22:49:15: rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
22:49:15: configsip: 0, SPIWP:0xee
22:49:15: clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
22:49:15: mode:DIO, clock div:2
22:49:15: load:0x3fff0030,len:1184
22:49:15: load:0x40078000,len:12784
22:49:15: load:0x40080400,len:3032
22:49:15: entry 0x400805e4
22:49:15: 
22:49:16: [    ESP] ESPixelStick v4.x-dev (Jan 13 2023 - 10:42:28)
22:49:16: [    ESP] v4.4.1-1-gb8050b365e
22:49:16: [FileMgr] Flash file system initialized. Used = 212992 out of 327680
22:49:17: [FileMgr] SD Card Size: 29656MB
22:49:17: [FileMgr] > System Volume Information
22:49:17: [FileMgr]   WPSettings.dat - 12
22:49:17: [FileMgr]   IndexerVolumeGuid - 76
22:49:17: [FileMgr] Configuration File: '/config.json' loaded.
22:49:17: [FileMgr] Configuration File: '/input_config.json' loaded.
22:49:17: [FileMgr] Configuration File: '/output_config.json' loaded.
22:49:17: Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
22:49:17: 
22:49:17: Core  1 register dump:
22:49:17: PC      : 0x40093090  PS      : 0x00060a33  A0      : 0x800938a5  A1      : 0x3ffb2330  
22:49:17: A2      : 0x3f800014  A3      : 0xffffffff  A4      : 0x3ffca56c  A5      : 0x00060823  
22:49:17: A6      : 0x00060820  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x0000000f  
22:49:17: A10     : 0x3f800794  A11     : 0x00000264  A12     : 0x00000016  A13     : 0x00060a23  
22:49:17: A14     : 0x00060a23  A15     : 0xfffffffc  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c  
22:49:17: EXCVADDR: 0x00000003  LBEG    : 0x4008c34d  LEND    : 0x4008c35d  LCOUNT  : 0xfffffffb  
22:49:17: 
22:49:17: 
22:49:17: Backtrace:0x4009308d:0x3ffb23300x400938a2:0x3ffb2350 0x40093af4:0x3ffb2370 0x40084672:0x3ffb2390 0x40084685:0x3ffb23c0 0x40084732:0x3ffb23e0 0x4013c691:0x3ffb2430 0x4013cda1:0x3ffb2450 0x4011b9ae:0x3ffb2480 0x40177707:0x3ffb24a0 0x401777e9:0x3ffb24e0 0x4018dfc3:0x3ffb2570 0x400f661b:0x3ffb2590 0x400d53ca:0x3ffb25b0 0x400d5cc7:0x3ffb2650 0x400ebe16:0x3ffb2710 0x400ea199:0x3ffb2780 0x400d069f:0x3ffb27a0 0x4010cc36:0x3ffb2820 
22:49:17: 
22:49:17: 
22:49:17: 
22:49:17: 
22:49:17: ELF file SHA256: 0000000000000000
22:49:17:

MartinMueller2003, 2023-01-28:
Please send me to log from the flash tool.

1. Use the flash tool to load the latest image.
2. Wait for the crash
3. On the flash tool, save the log to a file
4. Attach the file on this conversation.

You did not mention any configuration settings y9ou had made. Please send an image of the device config screen.

Attach a copy of the configuration to this issue:

1. Once the device is in the crash loop, connect it to the flashtool
2. On the flash tool, click the download configuration button
3. Zip up the downloaded configuration directory and post it in this conversation.

pakeonoahu, 2023-01-29:
[debuglog and config.zip](https://github.com/forkineye/ESPixelStick/files/10528862/debuglog.and.config.zip)

Zip contains the screenshot of the device setup before I click "Save Changes", saved logfile and the download of configs, when it's in the reboot loop, the only way I can get out of it is by flashing the firmware again, I found out as soon as I enable output 4 will cause the crash, doesn't matter what the settings were for output 1 to 3.  Thanks.

MartinMueller2003, 2023-02-23:
Sorry for the delay. I have been working on a flickering issue when RMT channels are used.

pakeonoahu, 2023-02-26:
no problem, I know you are busy and thanks for doing all this for the community.

MartinMueller2003, 2023-03-02:
I get a boot loop as soon as the ESP tries to connect to the network when no credentials are available. There is an error thrown in the DHP setting of the ESP libraries.

pakeonoahu, 2023-03-05:
DHP?(DHCP?) DHCP works fine, I tried it on four different esp32-cam boards, they acted the same way as soon as output 4 is checked and saved, could it be something got over written if that was selected?

MartinMueller2003, 2023-03-05:
I dug a bit deeper. The call to set DHCP is the same call that starts the WiFi hardware. Both my cam boards die at that point. Tried to load one of the examples from espresif and it died at the same point. There is somethng very wrong with my boards. I uploaded a new dist.zip that changes the way the outputs are scheduled. please give it a try.

MartinMueller2003, 2023-03-05:
Here is a link to the directory where I keep an updated dist.zip
https://drive.google.com/drive/folders/1bGrkXSMoGjWtX5iWspPFRogM2G6c1VXu?usp=sharing

pakeonoahu, 2023-03-06:
I just tried it, same behavior, I have the source from last year on 4.0-beta4, compiled it and that one didn't have the issue, but I couldn't figure out the date on that source code.  I wonder what has changed between then and now to cause the issue.  Here's the top of the Changelog:

# Changelog

### 4.0-beta4

- Moved FPP Remote to secondary input [Issue #388](https://github.com/forkineye/ESPixelStick/issues/388)
- Added option to change E1.31 port number [Issue #320](https://github.com/forkineye/ESPixelStick/issues/320)
- Display SD card stats on file mangement page [Issue #399](https://github.com/forkineye/ESPixelStick/issues/399)
- Fixed issue not being able to use GPIO0 as a relay pin [Issue #375](https://github.com/forkineye/ESPixelStick/issues/375)
- FPP multisync improvements
- Increased ESP8266 pixel limit to 1200 pixels
- Fixed issue with installing from 32bit Windows systems
- Added ESP32-CAM build enviornment and binaries
- Added RF calibration at boot and maximum TX limits to ESP8266 to help with stability
- Switched LWIP2 build configuration to "High Bandwidth, No Features"
- Switched to "yubox" fork of ESPAsnycWebserver
- Added PWM support for relays
- Added factory reset confirmation dialog

pakeonoahu, 2023-03-06:
I don't think anything was wrong with your ESP32-CAM, I built the latest source, it reboot loop starts at DHCP. 
[FileMgr] Configuration File: '/config.json' loaded.
[FileMgr] Configuration File: '/input_config.json' loaded.
[FileMgr] Configuration File: '/output_config.json' loaded.
⸮⸮ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12784
load:0x40080400,len:3032
entry 0x400805e4

MartinMueller2003, 2023-03-06:
The crash is happening someplace in the WiFi Startup function in the WiFi library. Need to figure out what changed in there.

MartinMueller2003, 2023-03-07:
The framework and the RTOS code have been updated. That has often had uninteresting side effects.

MartinMueller2003, 2023-03-08:
Had to go back to an earlier PIO version to get a stable build. 

Confirmed, enabling port 4 causes the system to crash. On reboot it tries to init port 4 again and crashes.

MartinMueller2003, 2023-03-08:
Fixed in PR #626 and in the published dist.zip file.

closed 2023-03-08 by MartinMueller2003