forkineye/ESPixelStick #617: ESPixelStick push button for playlist

mc-sketchy, 2023-02-05:
### Discussed in https://github.com/forkineye/ESPixelStick/discussions/615

<div type='discussions-op-text'>

<sup>Originally posted by **mc-sketchy** February  5, 2023</sup>
is it possible to add a push button to espixelstick that would cycle through the fseq files on the sd card? or perhaps go to next playlist item on a playlist?</div>

**_Describe what you are trying to accomplish_** 
I would like to add a push button functionality to existing ESPixelSticks (meaning not create a new board) that would scroll through fseq files stored on the sd card. The end goal is a stand alone light prop that once configured can be controlled with the button and does not require a wifi connection to be used. Ideally, the user would be able to turn prop on/off with a long press, and choose the desired effect with normal presses. 

**_How you would like it to work_**
1. This should work similar to how buttons are currently programmed in FPP
2. Playlist and fseq files added to sd card
3. Button connected to GPIO header
4. Long press and short press Event(s) assigned to GPIO in the Device Setup Page
	- Long press assigned power on/off
	- Short press assigned advance to next playlist item

MartinMueller2003, 2023-05-20:
Still working on this. At the moment I have effects turning on/off with a long push and moving to the next effect on a short push. Now that the infrastructure is in place, I will add this to the fseq palyer.

mc-sketchy, 2023-05-21:
Great.  I look forward to it. I appreciate your time and work.

MartinMueller2003, 2023-05-23:
You can give it a try using this image: https://drive.google.com/file/d/1ihIKCxoccCDYFvfUOL_RCftpTc7yDxYH/view?usp=share_link

MartinMueller2003, 2023-05-24:
Created PR.

MartinMueller2003, 2023-06-06:
Integrated into main

closed 2023-06-06 by MartinMueller2003

mc-sketchy, 2023-06-15:
apologies Martin. I have been super busy lately and just seeing your posts about this.  I will try it out this weekend.  Thank you so much for your work!