forkineye/ESPixelStick #618: Wt32-etho1   can't get it flashed with ESPixelStick tried all suggestion below  able to flash WLED and esphome

stonecreek2007, 2023-02-07:
### Discussed in https://github.com/forkineye/ESPixelStick/discussions/605

<div type='discussions-op-text'>

<sup>Originally posted by **Francesco-Rugg** December 12, 2022</sup>
Can I test [ESPixelStick v4.0-beta4](https://github.com/forkineye/ESPixelStick/releases/tag/v4.0-beta4)
For WT32-ETH01?</div>

MartinMueller2003, 2023-02-13:
Yes that is a supported platform. What issue are you having getting it installed?

stonecreek2007, 2023-02-13:
The issue is that it doesn't see the wt32etho board gives the error message
of not connected but it is. I have tried to do the three times io1 to
ground reset but still the same error message.
But then I try wled or esphome with the same flashing tool and they flash
with no problem. I took this wt32etho off a (Wasatch advanced eight) that
runs wled and thought maybe I could get ESpixelstick flash on it because it
works better with FPP.

Thanks for your response
Sol

MartinMueller2003, 2023-03-09:
I just used flashtool to program my wt32-eth01 board. it is as ugly as I remember it being a year ago. Here are the steps:

1. Connect the UART to the device
2. Connect a wire between gpio0 and ground. Make sure the wire stays in place.
4. On the flash tool, press the flash device button.
5. When the words "COM 3" (or whatever com port you are using) appears, apply power to the device.
6. When the words "COM 3" (or whatever com port you are using) appears, Quickly remove and reapply power to the device. Do this anytime you see the COM message appear. For me, this was three times.