forkineye/ESPixelStick #619: Support for DMX input?

Jredbart, 2023-02-12:
Hi there,

I have come across this project and the ESPixelStick V3 board which I've noticed includes a MAX485 transceiver for DMX output.

I was wondering if there is any support for DMX within the firmware as I don't see any hardware limitations as to why this isn't possible. I have searched the Wiki and the guides on forkineye website but am unable to find any answer.

Kind regards.

MartinMueller2003, 2023-02-13:
Both the V3 and the V4 versions of the software support DMX output. V3 requires  a special build. V4 has it integrated into the main image.

Jredbart, 2023-02-13:
Hi Martin, 

Thank you for your response. However, I wanted to know if you can provide a wired DMX input into the pixel stick rather than output DMX from the pixel stick. 

Kind regards

MartinMueller2003, 2023-02-13:
no, the RS-485 interface is set up for output only. What would you use this for?

Jredbart, 2023-02-13:
I have a situation where wifi is prohibited and I want to be able to control my pixel tape via my dmx software. I know there’s E1.31, SACN protocols but I don’t carry router / switch or any networking equipment and there is no Ethernet on the boards. 

I wanted to use wired DMX to be able to control interface with the pixel stick as I would over a network. (Was going to have separate university’s for all pixels)

MartinMueller2003, 2023-02-13:
There are plenty of ESP32 based platforms with Ethernet interfaces in the list of supported devices. All you need is an Ethernet cable from the ESP to the device that can generate E1.31 data. Use static IP configuration at both ends of the connection and you are good to go.