forkineye/ESPixelStick #622: Add support for Intertek (American Lighting / Better Homes and Gardens) outdoor pixels

forkineye, 2023-02-24:
This is to add support for 24V Intertek outdoor flex lights which are sold by Walmart and Costco under their own brand names.  So far, I have come across two versions of these lights. Both follow the ws2811 protocol very closely, but with padding between each 24bit chunk of pixel data and very long reset / latch times.  For the newer models of these lights, the padding and reset is much larger and seems to be required.  The older models of these lights will work with standard ws2811 timing, however the newer ones will not.  [Here](https://github.com/forkineye/ESPixelStick/files/10825728/New_Intertek.zip) are the DSview .dsl and exported .vcd files from a capture of the newer model controller.  The pixel bit timing is the same as ws2811, however the following discrepancies of the newer lights were noticed:
Inter-pixel padding of 20us pulled low
Reset / Latch time of 12.5ms

Newer models that do not work with ws2811:
https://www.walmart.com/ip/Better-Homes-Gardens-Color-Changing-Chasing-Neon-Flex-Light-16-4-feet/752676062
https://www.costco.com/ProductDisplay?catalogId=10701&partNumber=100788981

At the time of purchase (2/2023), this reseller had the older model which works with ws2811:
https://www.amazon.com/gp/product/B097HY6NF3

MartinMueller2003, 2023-02-24:
Did you try to adjust the inter frame gap on the WS2811 settings page? If IFG is the only difference then that should make it work.

forkineye, 2023-02-24:
> Did you try to adjust the inter frame gap on the WS2811 settings page? If IFG is the only difference then that should make it work.

I did and no joy. I can't work on it this weekend, was going to start on it next week.

MartinMueller2003, 2023-02-24:
I was thinking about it and the easiest way is to derive a new class from the WS2811 class. The only real function is to change the default IFG set at instantiation to the new desired default. Then create the UART and RMT variants and you are done. First I would check the output waveform to see why the IFG change did not work.

forkineye, 2023-04-04:
Closing this as I confirmed today it's a hardware issue. Standard timing was able to work.

closed 2023-04-04 by forkineye