forkineye/ESPixelStick #624: Dig Octa Brainboard Flickering/Stuttering

daklights, 2023-02-25:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
EspixelStick 4.0-ci4104030400
Build: Feb 6, 2023 13:21:42

**Hardware Version**
esp32_quinled_dig_octa

**Binary release or compiled yourself?**
Binary download

**Operating System (and version)**
Windows 10 22H2 (Build 19045.2604)

**Web Browser (and version)**
Chrome 110.0.5481.104

**Access Point**
N/A (hardwired via ethernet port)

**Describe the bug**
Two issues at play here, looks like one may have not been reported as an issue but might already be in the works to resolve.

1) Stuttering: when accessing the "Home" screen while DDP is being sent, the display updates approximately once per second (the DDP Status fieldset, showing total packets, total kbytes, errors).  Every time this fieldset updates, the WS2811 light outputs (all 8 of them) temporarily pause/freeze for about 1/4 of a second.  They then catch up, but pause again when the DDP metrics on screen update again.  When I close the browser, or when I move to another screen in the browser, like the "Network Setup" or "Device Setup" screens, the pause/freeze does not occur.  This happens every time the "Home" screen is visible and DDP Status fieldset is updating.  I've rebooted the device thru admin, power cycled it completely, nothing seems to make a difference is the "Home" tab is visible in a browser.

2) Flickering: there is some randomly occurring flickering on single outputs.  I cannot consistently predict or reproduce this, but when all 8 ports are being used, one single output will flicker a different color across the strand.  For example, I have one output set to slowly fade through all 100 pixels with red coloring.  Occasionally, this strand will briefly flicker green through all 100 pixels.  Other times, it will flicker off all pixels before picking back up with the red.  I've confirmed this is not related to the pixels themselves (swapped out for other pixel strands) and not related to the pigtail/cabling (I can move these same cords to an espixelstick and the flickering does not occur).  It seems to be solely related to the Dig Octa Brainboard.  Unfortunately, I don't have any other Dig-brand controllers to test with, just the brainboard and an espixelstick in terms of hardware.  I will note that there seems to be a pull request (#623) for this issue by @MartinMueller2003, though I'm not smart enough to know if this is related or just happens to be talking about 8 port flickers in a different way.

MartinMueller2003, 2023-02-25:
The issue I have been working on (dig quad) uses the same CPU configuration as the octa. The pull request I have pending fixes both the issues you are seeing. They have the same root cause, too much time spent in the ISRs.

daklights, 2023-02-25:
Awesome, thanks.  I'm happy to help test any updates if needed!

MartinMueller2003, 2023-02-25:
Take a look at the dist found here:
https://drive.google.com/file/d/1ihIKCxoccCDYFvfUOL_RCftpTc7yDxYH/view?usp=share_link

daklights, 2023-02-25:
I just re-flashed the Dig Octa brainboard and tested with mixed results:

1. The stuttering when "Home" screen is visible in browser (specifically the DDP Status fieldset updates) is no longer present.  I can safely display the "Home" screen and there's no visible impact to the WS2811 outputs when the DDP Status information updates regularly about once per second.  Seems this is resolved!

2. The flickering is still very present, unfortunately.  It happens at random, with no discernable pattern or consistency, and happens to all outputs on the Dig Octa brainboard... not appearing to be isolated to a specific output or set of outputs (1-4 or 5-8).

3. A new side effect of the distro you had me try/install today (4.x-dev Build Date: Feb 24, 2023 17:44:32): if Output 9 is toggled to "Relay" instead of "Disabled" state, then all eight WS2811 outputs "jitter" (best term I can use to explain the situation) at a consistent frequency... it's almost like the relay configuration's PWM settings are affecting the WS2811 outputs, though I don't understand enough about that to make anything more than a slight suggestion.  When I toggle Output 9 from "Relay" to "Disabled" state, the Dig Octa saves the configuration change and then seems to crash (the browser loses WebSocket connection and never comes back and only a hard power off/on reset brings it back online).  When the unit comes back on, though, the "jittering" is not present.  If I toggle Output 9 from "Disabled" back to "Relay"... the Dig Octa crashes again, but when it comes back online after a hard power off/on reset, the settings I'd submitted before crashing were saved and the "jittering" is present again.  This behavior is not present in the previous version of beta4 firmware I was running last night (EspixelStick 4.0-ci4104030400 Build: Feb 6, 2023 13:21:42).  I tried to capture a video of this "jittering" effect, but my cell phone's camera refresh rate must not be sufficient enough to detect the jittering.  When comparing the Dig Octa brainboard's outputs alongside an espixelstick's output, though, the jittering is very noticeable.

MartinMueller2003, 2023-02-25:
Thanks for testing. Looks like I need to pull out the octa and see what is going on. More clutter on my desk.

MartinMueller2003, 2023-02-26:
I just uploaded a new version. The relay issue was caused by an artificial 50ms delay that messed up the pixel outputs. 
The flickering was caused by using uart and rmt driven channels at the same time. Latest changes makes that a bad choice for the 8 port boards. 
This image should be better. Not sure if all of the issues are resolved. I am still working on 8 port issues reported by other users.

daklights, 2023-02-26:
I see definite improvement with this latest version (4.x-dev Build Date: Feb 25, 2023 18:51:17), but don't think things are out of the woods just yet.

1. The stuttering when "Home" screen is visible in browser (specifically the DDP Status fieldset updates) is still no longer present when running DDP... but I got my show FPP instance back out of storage, and unfortunately, when FPP is pushing to the Dig Octa brainboard as an FPP Remote, the stutter has returned when the "Home" screen is visible in browser.  I can only assume that the "FPP Remote Status" fieldset updates each second are causing the stutter this time around.

2. The flickering is completely gone from all 8 ports when running DDP, but unfortunately, continues to rear its ugly head when running FPP Remote.

3. The jittering issue that occurred when Output 9 was configured to "Relay" is now no longer present.  However, a strange new issue popped up in its place.  When Output 9 is configured to "Relay" (and thus, configured to GPIO33), Output 1 on the Dig Octa brainboard does not work.  The pixels are still attached and powered, but are getting no data signal it seems.  As soon as I toggle Output 9 from "Relay" to "Disabled" and save, Output 1 resumes normal operation... no other changes occur.  I will say that the Dig Octa brainboard still crashes anytime a config change is made to the Output 9 or Relay settings, but a quick hard power off/on reset cycle does the trick to get things back online with the saved config settings.

MartinMueller2003, 2023-02-26:
New version uploaded to take care of item 3. It turns out that the default port assigned to the relay outputs was the same as that used by pixel port 1 (GPIO 0). I have changed the default to -1 and added code to handle that.

As for issues that appear while the UI is up, we have long given up trying to make that work. If there are no issues when the UI is not up then we are declaring victory. 

I will look into the FPP remote issues this week.

MartinMueller2003, 2023-02-26:
On the version that I posted, I am not seeing any issues with FPP playback.

MartinMueller2003, 2023-02-27:
Fixes for these issues are in the latest pull request.

MartinMueller2003, 2023-03-02:
No longer seeing a flicker but we are now seeing a random RMT channel stop sending data. There is also a negative interaction between UART and RMT processing causing the output speed to degrade.

MartinMueller2003, 2023-03-08:
This is the same as the Dig Quad using 8 ports issue that has been resolved.

closed 2023-03-08 by MartinMueller2003