forkineye/ESPixelStick #628: V4 beta 5 - is there a newest firmware available?

roboengr, 2023-03-29:
**ESPixelStick Firmware Version:** v4 beta 4
**Hardware Version:** D1 mini 8266(esp8266MOD chip) with one relay output
**Binary release** used.
**Operating System (and version)** win 10 latest
**Web Browser** (and version);chrome latest
**Access Point:** Ubiquiti NanoStation locoM2 
**Describe the bug:** crash and status settings are opposite,  turn off dynamic and it loads dynamic.  turn on dynamic  it load static.
The loader jar seems to be defective.  I am trying to use the firmware just the way it is down loaded.  it seems to be not selecting the d1 mini 8266 for some reason.

tried all betas with different results from each all not working. there was talk about relay fixes GPIO -1 compared to zero. and other modifications.  should there not be a new version out now since nov 21 2022?  where do we get the latest version?

roboengr, 2023-03-29:
[image]

roboengr, 2023-03-29:
will the v4 beta 4 not work on this device?

forkineye, 2023-03-29:
You can grab the latest CI builds - https://github.com/forkineye/ESPixelStick/actions/workflows/build.yaml.  Just click on the newest entry and scroll down to "Release Archive" under "Artifacts".

MartinMueller2003, 2023-04-13:
The latest V4 builds support multiple ESP8266 platforms as well as a rich set of ESP32 platforms. It even has a version for ESP32 D1 Mini running on the ESPv3 hardware. Please take a look at the supported platforms list.