forkineye/ESPixelStick #630: Effect - Marquee with Flashes

MartinMueller2003, 2023-05-08:
Please create a new effect that allows the user to simulate a classic Marquee on a string of lights.

This is similar to a chase with two colors and three pixels to a group. The groups move one pixel forward at a time to create the moving chase effect.

There should be a strobe effect that repeats at a given interval over the time the effect is running

The effect is to be started / stopped using an external trigger (aka button)

MEM - The above is from notes taken from an ESP user. The below is an expansion on the feature to make it more generic and useful in a wider setting.

- Groups
    - Allow` the user to configure the number of groups to chase
    - Each group can be a different color
    - Each group can be a different size (aka number of pixels in the group)
    - The intensity of the pixels in the group must be able to support a fade in or a fade out effect
- The speed of the chase must be configurable
- Flash:
    - The flash duration must be configurable
    - The flash rate must be configurable
    - Flash rate may be set to "Random". If random then min/max delay times must be set.

The effect will be controlled as the other effects are controlled. There is another ticket open asking to be able to trigger effects using an external trigger. This effect will take advantage of that functionality to meet the button requirement.
-

MartinMueller2003, 2023-05-16:
Making progress. Marquee and flashing is working and will be submitted for integration. Triggering is still in progress.

MartinMueller2003, 2023-05-16:
PR: Added Marquee and Random Flash support. #632

PtitBry, 2023-05-18:
Hi, Many thanks for your help ;)

MartinMueller2003, 2023-05-23:
You can give it a try using this image:
https://drive.google.com/file/d/1ihIKCxoccCDYFvfUOL_RCftpTc7yDxYH/view?usp=share_link

MartinMueller2003, 2023-05-24:
The GPIO used depends on your platform. The pin is configurable as is the polarity and definition of a long pull. 

By default, pulling the assigned pin low for less than 2 seconds is a short push and over 2s is a long push.  You also need to enable the feature via the UI before it does anything.

MartinMueller2003, 2023-05-24:
Here is a picture of my test setup
[image]

PtitBry, 2023-05-24:
I have a "node MCU V3", I think an ESP V3 (ESP 8266MOD).

So I can put the boton on any pin D0 to D8, and the data of the WS2811 strip led ?
Vin : 5Vcc
GND : gnd

MartinMueller2003, 2023-05-24:
PR Created.

MartinMueller2003, 2023-06-06:
Integrated into main

closed 2023-06-06 by MartinMueller2003

PtitBry, 2023-06-11:
> Integrated into main

Hi, Many Thanks Martin 😁