forkineye/ESPixelStick #636: Error when using internal clock for Lan8720

processor83, 2023-06-09:
**ESPixelStick Firmware Version**
4.x-dev compiled today

**Hardware Version**
my board 
GPIO_Defs_ESP32_ESP3DEUXQuattro_DMX

**Binary release or compiled yourself?**
compiled myself and older version provided for test..

**Operating System (and version)**
Windows 10

**Web Browser (and version)**
Chrome

**Access Point**
WIRED

**Describe the bug**

When external ossciator is not used there is errors because gpio is defined as -1

// Pin# of the enable signal for the external crystal oscillator (-1 to disable for internal APLL source)
#define DEFAULT_ETH_POWER_PIN          gpio_num_t(-1)
#define DEFAULT_ETH_POWER_PIN_ACTIVE   HIGH

UI is not happy and show a red exclamation mark and if i check the serial outbut i have this error.

[ 10694][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (10687) gpio: gpio_set_level(226): GPIO output gpio_num error

MartinMueller2003, 2023-06-13:
If you change the clock mode, then you need to specify which pin the power signal is going out. The error you mention indicates you failed to do that. It is my understanding that using the external clock requires the ability to power cycle the interface.

processor83, 2023-06-14:
is it set like this and i get exclamation mark in UI nest to the -1 pin... also web interface is not accessible via ethernet interface  juste afther i acces via wifi then i can acces via ethernet..  

I dont know if its only a web ui bug for the -1 pin and a completly different bug for the part of the web ui not accesible via ethernet before i connect to the wifi  even for 1 sec just connect and disconnect no need to acces 192.168.4.1

#define DEFAULT_ETH_CLK_MODE           eth_clock_mode_t::ETH_CLOCK_GPIO17_OUT

// Pin# of the enable signal for the external crystal oscillator (-1 to disable for internal APLL source)
#define DEFAULT_ETH_POWER_PIN          gpio_num_t(-1)
#define DEFAULT_ETH_POWER_PIN_ACTIVE   HIGH

// Type of the Ethernet PHY (LAN8720 or TLK110)
#define DEFAULT_ETH_TYPE               eth_phy_type_t::ETH_PHY_LAN8720

[ 10654][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (10647) gpio: gpio_set_level(226): GPIO output gpio_num error

MartinMueller2003, 2023-08-03:
Is there a reason you are using the internal clock? It is usually a better idea to let the LAN device drive the clock. The clocks for Ethernet are pretty tricky,

MartinMueller2003, 2023-08-17:
Please retest using the build from today. There was an issue in the base OS that was causing massive data loss due to clocking issues. I have moved the version backwards in time to a version that works with the existing implementations.