forkineye/ESPixelStick #637: Web interface does not work afther a reboot

processor83, 2023-06-09:
**ESPixelStick Firmware Version**
4.x-dev compiled today

**Hardware Version**
my board 
GPIO_Defs_ESP32_ESP3DEUXQuattro_DMX

**Binary release or compiled yourself?**
compiled myself and older version provided for test..

**Operating System (and version)**
Windows 10

**Web Browser (and version)**
Chrome

**Access Point**
WIRED

**Describe the bug**

I have a static ip set,
Allow wifi is not checked.
I can ping the static ip i set. 
---------------------------------------------------
Web interface is not accesible.
---------------------------------------------------
Then i connect to the fallback wifi network

Now i can acces the web interface

I this state art-net is received but no output .. and there is error in serial output 

POLL from 192.168.0.144 broadcast addr: 10.0.1.255
[138967][E][WiFiUdp.cpp:185] endPacket(): could not send data: 118
POLL from 192.168.0.144 broadcast addr: 10.0.1.255
[143968][E][WiFiUdp.cpp:185] endPacket(): could not send data: 118
POLL from 192.168.0.144 broadcast addr: 10.0.1.255
[148966][E][WiFiUdp.cpp:185] endPacket(): could not send data: 118
POLL from 192.168.0.144 broadcast addr: 10.0.1.255
[153962][E][WiFiUdp.cpp:185] endPacket(): could not send data: 118
POLL from 192.168.0.144 broadcast addr: 10.0.1.255
[158966][E][WiFiUdp.cpp:185] endPacket(): could not send data: 118
POLL from 192.168.0.144 broadcast addr: 10.0.1.255
[163966][E][WiFiUdp.cpp:185] endPacket(): could not send data: 118

----------------------------------------------------------------------------

I can disconnect from wifi and web interface stay accesible.

I this state art-net is received but no output ..

------------------------------------------------------------------------------

i set Secondary Input as effect save/reboot connect to wifi fall back ap and now art-net work....

i remove effect as the Secondary Input save and reboot now art-net stop working.

I know that is is not super clear but there is very weird situations....

i can acces web interface over ethernet but ethernet status connect = no i disconnect and reconnect the cable now its YES and art-net work...

MartinMueller2003, 2023-08-03:
Please retry using this version:

https://drive.google.com/drive/folders/1bGrkXSMoGjWtX5iWspPFRogM2G6c1VXu?usp=drive_link

MartinMueller2003, 2023-08-12:
Confirmed.

MartinMueller2003, 2023-08-17:
dist.zip has been updated with this change.

MartinMueller2003, 2023-08-17:
Changes have ben merged to main.

closed 2023-08-17 by MartinMueller2003