forkineye/ESPixelStick #638: PixelStick stays in AP mode and does not try to reconnect upon wifi connect failure

garymueller, 2023-06-29:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
master

**Hardware Version**
D1 mini32 - uploading Quinled Quad build

**Binary release or compiled yourself?**
binary release
(bug and fix identified with a compiled version) 

**Operating System (and version)**
Windows 11

**Web Browser (and version)**
Google Chrome

**Access Point**
Esp always opens an access point upon wifi connect failure

**Describe the bug**
The firmware always opens up an accesspoint upon wifi connect failure.  This is due to the firmware always setting the "StayInApMode" as on.  This is an issue because upon powerfailure, you will have to log into each pixelstick and reboot it after the real ap is back online. 

The bug is at  script.js line 1188
`wifi.StayInApMode = $('#network #wifi #StayInApMode').val();`

The issue is that the checkbox is treated as a value rather than a boolean.  The code should be
`wifi.StayInApMode = $('#network #wifi #StayInApMode').prop('checked');`

My test indicates that the above change resolved my issue.  The pixelstick would go into a reboot cycle until the real AP was back online which was the desired (and intended) behavior.

garymueller, 2023-06-29:
Wow. Thanks for the quick turn around and all your contributions to this project.  Do you want me to close the issue?

MartinMueller2003, 2023-06-30:
yes,please close the issue

closed 2023-06-30 by garymueller