forkineye/ESPixelStick #643: EspPixelstick UI goes unresponsive, You can connect once and that is it

patdelaney, 2023-07-26:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
The last view builds. The one I just tested was the ci build from 7/24/2023

**Hardware Version**
ESPPixelstick V3  or Artstick with SD card

**Binary release or compiled yourself?**
CI build 7/24/2023

**Operating System (and version)**

**Web Browser (and version)**
Chrome Version 114.0.5735.248 (Official Build) (64-bit)

**Access Point**

**Describe the bug**
Ever since I got back from Christmas Expo 7/172023 I have been having issues navigating the pixelstick UI. You may be able to connect once, for a bit, then pixelstick spontaneously reboots. When it comes back the webui is unresponsive, but you can ping the stick.
The sticks I was using were fine at expo. I suspect a chrome update. 
When I tested with a new laptop and tried to connect via chrome, it downloaded a file. I will attach to ticket.

On another one when I clicked on diagnostics, it downloaded a file and the stick is unresponsive
[image]

it never appears to finish the download

[Unconfirmed 158206.zip](https://github.com/forkineye/ESPixelStick/files/12166650/Unconfirmed.158206.zip)

MartinMueller2003, 2023-07-26:
Is this using the original ESP8266 mini Processor board?

MartinMueller2003, 2023-07-26:
Been running on the Diags page showing a random effect output on 100 pixels for the past few hours. Have switch pages on occasion and everything is working fine.

MartinMueller2003, 2023-07-26:
So I just kept clicking around and ended up in a situation where the esps.css file was somehow corrupted. It looks like something happened to the FileSystem.

bloojoop, 2023-07-26:
I was able to reproduce what Pat is seeing.  I am using build from July 25 version ESPixelStick_Firmware-4.0-ci5659517986.   I have a esplixelstick v3 hardware with a D1 mini.  I can flash it fine, and the web ui is still responsive.. I can reboot it just fine and still seems responsive.  I then configured input 1 as DDP, and saved.  It was fine until I rebooted, once I rebooted after configuring the first input, then it would also hang up and not respond.  Here are my log files from the various reboots while the flasher was still connected. 

[ui-hang.log](https://github.com/forkineye/ESPixelStick/files/12178027/ui-hang.log)

MartinMueller2003, 2023-07-26:
DDP is the key. DDP on and a reboot (which the ESP8266 does a lot) results in no UI. No sure why but this is something that can be fixed

MartinMueller2003, 2023-07-27:
The UI fails on this error: 

Error with Permissions-Policy header: Origin trial controlled feature not enabled: 'browsing-topics'.

bloojoop, 2023-07-27:
Hm, Likely same issue but does not seem to be limited to DDP... I just configured ws2811 output with no inputs... and changed universe size to 510... saved, was fine until reboot.. Now that I rebooted it is hanging the same way... Likely the same underlying issue just wanted to report that it seems to not be limited to choosing DDP as an input.

MartinMueller2003, 2023-07-27:
When development mde is in use, everything works. When everything is compressed and munged things stop working

patdelaney, 2023-07-28:
any chance you are running into a memory or or buffer limitation when
compressing and decompressing?

Pat

MartinMueller2003, 2023-07-28:
that is something i am looking at

MartinMueller2003, 2023-07-28:
Still digging into this. It looks like there is a problem serving the esps.js and esps.css files. There is no V3 code involved in this operation. It is handled entirely by the web server library.

MartinMueller2003, 2023-07-28:
> any chance you are running into a memory or or buffer limitation when compressing and decompressing? Pat
> […](#)

Nope. Not even getting to the point where we use the web socket. This is just generic web hosting.

patdelaney, 2023-07-28:
I just tried the CI build on esp8266 and got the same issue again
[image]

MartinMueller2003, 2023-07-28:
I have to create a pull request and Shelby has to review it and then the CI builds will get kicked off to get you thos images.

MartinMueller2003, 2023-07-28:
New dist.zip is at https://drive.google.com/drive/folders/1bGrkXSMoGjWtX5iWspPFRogM2G6c1VXu?usp=sharing

MartinMueller2003, 2023-07-28:
Fixed in 645

forkineye, 2023-07-28:
It's merged, thanks Martin. I've been completely slammed with work lately.

patdelaney, 2023-07-29:
Sorry it's still broken

patdelaney, 2023-07-29:
I tried Martins build on a different PC and it works. I'll try clearing the browser cache

bloojoop, 2023-07-29:
I download and flashed the lateset build ( FW Version4.0-ci5696732341Build DateJul 28 2023 - 22:17:29 ) and configured ddp for primary, fpp remote for secondary, and 2811 for output... so far it appears fairly stable for me.

patdelaney, 2023-07-29:
try connecting from a different browser at the same time

patdelaney, 2023-07-29:
I may be headed into the weeds on this, but is it possible that when flashing a new image, it doesn't completely wipe a formerly bad version? Could something be left behind. Does the flash tool erase/wipe/zero out everything before writing the new image?

patdelaney, 2023-07-29:
it's still very unstable IMHO. I am currently testing with an Artstick that has SD card I test with and without the card and also just with the esp8266 stand alone and get unpredictable results.

MartinMueller2003, 2023-07-29:
> I may be headed into the weeds on this, but is it possible that when flashing a new image, it doesn't completely wipe a formerly bad version? Could something be left behind. Does the flash tool erase/wipe/zero out everything before writing the new image?

The flash tool does a complete erase of all flash memory (at least it tries to do that) and then verifies that all memory locations have been reset.

MartinMueller2003, 2023-07-29:
I dont know about the CI build, but the dist image I sent a link to is working just fine. I just flashed an old device and it is working fine. Keep in mind that the ESP8266 is not very stable when the UI is getting a lot of use. It will reboot regularly due to internal memory leaks in the libs we are using. But it always comes back.

MartinMueller2003, 2023-07-29:
You may be right. I just spent a few minutes clicking from one menu to another until the device crashed. After reboot it again failed to load esps.js. This is getting annoying.

MartinMueller2003, 2023-07-29:
Broke up the esps.js file into its parts. Consistently fails to load on a js file but not a specific file.

patdelaney, 2023-07-29:
baby steps but at least I am not crazy

MartinMueller2003, 2023-07-29:
I put in a heap monitor. The moment th ui tries to connect, the available heap drops to less than 3000 bytes. that is too small to support an http connection and the transfer of js files halts.

MartinMueller2003, 2023-07-30:
More data. Even if you get the number of concurrent connections down to a point where the heap is not exhausted, the Async Web server refuses to send data whe the heap is under 10K. So you get a connection but no data phase. Need to find out if the trip level is adjustable and also find out if we can free up more ram.

patdelaney, 2023-07-31:
Since I'm not sure what we could remove to save on "ram", let me toss thses ideas out.
What is we remove some of the edge case protocals, Would that save RAM?
If it would, How much work would it be to spin protol specific builds? Maybe 1 build for 2811 and GECE, and then a second build for Serial DMX and Renard?

cybercop23, 2023-07-31:
FWIW I tested the latest one on 2 identical ESP32s/AE+ Quindor/DrZzzs special ESP32 with the AE+ hat to get 3 more ports and the SD card. One works without any issues, the 2nd one experiences the same issues as Pat. All I did was turn on Outputs 1&2 and set first 5 to WS2811

MartinMueller2003, 2023-08-03:
try this version
https://drive.google.com/drive/folders/1bGrkXSMoGjWtX5iWspPFRogM2G6c1VXu?usp=drive_link

patdelaney, 2023-08-03:
I am at the office. I'll check tonight

bloojoop, 2023-08-03:
I loaded todays version on a d1 mini.  Just using it powered up via the usb cable.  network appears to be very unstable.   attached is a just a continuous ping showing mass packet loss.   I could not do anything the browser interface. 

[aug3rd ping.txt](https://github.com/forkineye/ESPixelStick/files/12255399/aug3rd.ping.txt)

cybercop23, 2023-08-03:
Ditto. I get a bare bone html page with a spinning loading page. looks like
the css hasn't been applied

patdelaney, 2023-08-03:
I am using the build martin Posted and It's connecting and occasionally I get reboots. But it comes back each time. This test is an artstick with no sd card installed. I saw this after configuring DDP in and ws2811 out:

18:29:15: [WebMgr] WS client connect - 2
18:29:16: [FileMgr] ***ERROR: Failed to allocate memory for the GetListOfSdFiles web request response.***
18:29:16: [FileMgr] ***ERROR: Failed to allocate memory for the GetListOfSdFiles web request response.***

cybercop23, 2023-08-04:
DugUno with ESP32 + AE+ hat with SD card. Before this one it was loading...
[image: image.png]

MartinMueller2003, 2023-08-04:
> I am using the build martin Posted and It's connecting and occasionally I get reboots. But it comes back each time. This test is an artstick with no sd card installed. I saw this after configuring [...]

Thanks Pat. Yes it is still pretty low on memory. Still looking to see where I can get more memory. May have to rework the Web Socket interface to get rid of the buffer.
That means changing a lot on the UI to make it more REST like.

patdelaney, 2023-08-04:
the Pixel stick I put your latest image on lst night, I set it for WLED rainbow and test let it run. Today I tried to openm the web UI and it hanging. But the Rainbow effect is still running.  I tried ruinning a postman WS call 
http://192.168.1.170/fppjson.php?command=getHostNameInfo
and it returned
{
    "HostName": "shelby-sd6",
    "HostDescription": "ESPixelStick"
}
than a few more retrys of the same call and it hangs

MartinMueller2003, 2023-08-11:
Still working on this. I had to find a bunch of memory. This caused me to look at the WebSocket server. It turns out the WS Server used 10KB of memory (buffers etc). I have now eliminated all WebSocket support and replaced all of the functionality with DTTP operations. All work on the ESP8266. The output_config download for ESP32 does not seem to complete. The last chunk seems to get lost.

patdelaney, 2023-08-12:
I downloaded last nights CI build and I am testing on Artstick with SD card. (currently no card in slot)
After I configured it the Web UI was very responsive and no reboots. At boot up the Heap was 24456. I then configured it for E1.21 input and WS2811 output 100 pixels 20% brightness.  Now Heap is at 23568. Saved the config and then noticed this in the serial output window. and it write it out every second:

07:56:33: [WebMgr] ERROR: Status Doc is too small
07:56:34: [WebMgr] ERROR: Status Doc is too small
07:56:34: [WebMgr] ERROR: Status Doc is too small

Note that there is no sd card in the slot  I'm guessing that is the issue

MartinMueller2003, 2023-08-12:
Nope. That means one of my constants is too low for the status response on the home page. That is an easy fix. and will be in an upcoming push to my fork. 

WARNING: Only ESP8266 is working. The ESP32 TCP stack has a bug that prevents the output config from getting saved.

MartinMueller2003, 2023-08-12:
New dist with ESP32 support and file upload / management

https://drive.google.com/drive/folders/1bGrkXSMoGjWtX5iWspPFRogM2G6c1VXu?usp=sharing

patdelaney, 2023-08-12:
Not reading contents of the SD card
[image]

But when you reboot the serial output can see the files
18:56:45: [FileMgr] SD Card Size: 14860MB
18:56:45: [FileMgr] > System Volume Information
18:56:45: [FileMgr]   IndexerVolumeGuid - 76
18:56:45: [FileMgr]   WPSettings.dat - 12
18:56:45: [FileMgr] FPP-Matrix_Demo.fseq - 108072
18:56:45: [FileMgr] pole_butterfly.fseq - 306072
18:56:45: [FileMgr] pole_random.fseq - 306072
18:56:45: [FileMgr] MyPlayList.pl - 998

MartinMueller2003, 2023-08-13:
And mine looks fine on both ESP8266 and ESP32
[image]

MartinMueller2003, 2023-08-13:
press F12 key and look at the console tab. It may be throwing errors that would help figure out what is going on.

patdelaney, 2023-08-13:
[image]

MartinMueller2003, 2023-08-13:
thanks, that means the list is empty. how many files are on the sd card?

patdelaney, 2023-08-13:
There are 4 files on the card. With the serial connected you can see when it boots up there are files on the card:

07:45:19: [FileMgr] SD Card Size: 14860MB
07:45:19: [FileMgr] > System Volume Information
07:45:19: [FileMgr]   IndexerVolumeGuid - 76
07:45:19: [FileMgr]   WPSettings.dat - 12
07:45:19: [FileMgr] FPP-Matrix_Demo.fseq - 108072
07:45:19: [FileMgr] pole_butterfly.fseq - 306072
07:45:19: [FileMgr] pole_random.fseq - 306072
07:45:19: [FileMgr] MyPlayList.pl - 998

I've got Skype active if you want t jump on there

cybercop23, 2023-08-13:
@MartinMueller2003 thank you for all you help. Loaded this new one on DigUno AE+ w/ card... and gui seems to respond much much much better than before. No more slow web sockets barbershop bar. Configured as DDP/5 outputs. Still okay.

@patdelaney can you please try to upload config from xlights? i have a diguno ae+ configured with esp 4x and 100 pixles/port/5ports... put 5 CCs each one one port...

2023-08-13 11:45:08,287 8554896896 log_base [DEBUG] RecalcStartChannels takes 230ms.
2023-08-13 11:45:08,287 8554896896 log_base [DEBUG] Connecting to websocket 192.168.3.123 /ws.
2023-08-13 11:45:08,292 8554896896 log_base [DEBUG]     Connected.
2023-08-13 11:45:08,793 8554896896 log_base [ERROR]      Failed to convert to web socket 404.
2023-08-13 11:45:08,794 8554896896 log_base [ERROR] Error connecting to ESPixelStick controller on 192.168.3.123.

MartinMueller2003, 2023-08-13:
xLights wont work. It needs to be updated. Uploaded a test version for Pat. You should see this:

------ GetListOfSdFiles:FileMgr.cpp:866: ResponseJsonDoc.size(): 4 ------
------ operator():WebMgr.cpp:257: Files: {"files":[{"name":"TuneToSign.fseq","date":1640342670,"length":1579452},{"name":"01 The Steamroller [Fresh Aire 8].fseq","date":1668878924,"length":36598254},{"name":"02.TSO_ChristmasCannon.fseq","date":1668880394,"length":7557483},{"name":"03.God Rest Ye Merry Gentlement.fseq","date":1668877876,"length":6108423},{"name":"04.Good King Joy.fseq","date":1668878024,"length":6694759},{"name":"05.I Saw Three Ships.fseq","date":1668878112,"length":2961457},{"name":"07.Christmass Eve Sarajevo.fseq","date":1668875640,"length":7646333},{"name":"calibration.fseq","date":1633206034,"length":789756},{"name":"DDPTest.fseq","date":1677442196,"length":480116},{"name":"FrostyTheSnowman.fseq","date":1668875800,"length":6975077},{"name":"MonsterMash.fseq","date":1668878246,"length":7042198},{"name":"Nutrocker.fseq","date":1668878442,"length":9748882},{"name":"PeaceOnEarthLittleDrummerBoy.fseq","date":1668878618,"length":6238721},{"name":"PlayDates.fseq","date":1668886734,"length":2369156},{"name":"Santa Claus Is Comin to Town.fseq","date":1669557132,"length":11642199},{"name":"SyncTest.fseq","date":1634572524,"length":4738252},{"name":"Thriller.fseq","date":1668879666,"length":12337132},{"name":"TSO - What Is Christmas.fseq","date":1668879914,"length":6757943},{"name":"TSO - Wizards In Winter (Instrumental).fseq","date":1668880246,"length":7176469}],"SdCardPresent":true,"totalBytes":15923150848,"usedBytes":145442062} ------

MartinMueller2003, 2023-08-13:
upload == dist.zip

patdelaney, 2023-08-13:
I reloaded the firmware, and I am testing on Artstick with SD card
UI shows this
[image: image.png]

But the Serial output see the data

[  FPPD] Listening on port 32320
17:20:01: ------ GetListOfSdFiles:FileMgr.cpp:866: ResponseJsonDoc.size():
4 ------
17:20:15: ------ operator():WebMgr.cpp:257: Files:
{"files":[{"name":"FPP-Matrix_Demo.fseq","date":1689243096,"length":108072},{"name":"pole_butterfly.fseq","date":1613400328,"length":306072},{"name":"pole_random.fseq","date":1613400344,"length":306072},{"name":"MyPlayList.pl","date":1613400440,"length":998}],"SdCardPresent":true,"totalBytes":15581839360,"usedBytes":721214}
------
17:20:15: ------ GetListOfSdFiles:FileMgr.cpp:866: ResponseJsonDoc.size():
4 ------
17:20:34: ------ operator():WebMgr.cpp:257: Files:
{"files":[{"name":"FPP-Matrix_Demo.fseq","date":1689243096,"length":108072},{"name":"pole_butterfly.fseq","date":1613400328,"length":306072},{"name":"pole_random.fseq","date":1613400344,"length":306072},{"name":"MyPlayList.pl","date":1613400440,"length":998}],"SdCardPresent":true,"totalBytes":15581839360,"usedBytes":721214}
------
17:20:34:

MartinMueller2003, 2023-08-13:
Some time this week we need to get on a call where I can look at the network payload and set a breakpoint on processing the data, That output indicates the esp has put the proper data into the http response and the issue is in the client processing.

closed 2023-08-14 by forkineye

bloojoop, 2023-08-23:
@MartinMueller2003 , I just got back and got a chance to test with the release archive from https://github.com/forkineye/ESPixelStick/actions/runs/5943922254..   This version seems very stable at this point.  I also configured it in the newly release xlights 2023.13 and was able to upload outputs, and also upload sequence to the sd card via fpp connect.... I just wanted to report that this version seems to be golden from my quick test... Thank you again very much for your attention on this and knocking it out of the park :)