forkineye/ESPixelStick #649: broken SD card support in recent V4 CI ?

marcobrianza, 2023-08-17:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
ESPixelStick_Firmware-4.0-ci5179361640 SD OK
ESPixelStick_Firmware-4.0-ci5843367015 files not listed
ESPixelStick_Firmware-4.0-ci5837201311 SD card not detected

**Hardware Version**
QUINLED DIG OCTA
sd card 2GB

**Binary release or compiled yourself?**
CI release

**Operating System (and version)**

**Web Browser (and version)**

**Access Point**

**Describe the bug**

Today I tried latest CI release ESPixelStick_Firmware-4.0-ci5837201311 with my newly arrived quinled octa
I don't see the file management menu

going back to a release I used some time ago ESPixelStick_Firmware-4.0-ci5179361640 i see the files on the SD card

dowloading a more recent release ESPixelStick_Firmware-4.0-ci5837201311  I can see the SD but not the files included

There are some recent mods the the files handling that may have broken the SD card support, what do you think?

in all the cases the SD pins are the default ones for dig octa and are the same

Ciao
Marco

forkineye, 2023-08-17:
There have been a few commits since that release, specifically for ESP32 stuff. Can you try the one that was generated today? https://github.com/forkineye/ESPixelStick/actions/workflows/build.yaml
cheers,
-shelby

marcobrianza, 2023-08-17:
ah ok 4.0-ci5890809791 works!
thank you

closed 2023-08-17 by marcobrianza