forkineye/ESPixelStick #650: V4 file management

cybercop23, 2023-08-22:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
FW Version4.x-dev
Build DateAug 16 2023 - 20:47:42
ArchitectureESP32

**Hardware Version**
Board Nameesp32_quinled_uno_ae_plus

**Binary release or compiled yourself?**
Released

**Operating System (and version)**
Win 11

**Web Browser (and version)**
Chrome

**Access Point**
No

**Describe the bug**
File management shows the files on the SD card, but when selecting and trying to remove one, the check box gets unchecked but the file remains.
No errors show up in the console.

MartinMueller2003, 2023-08-22:
It looks like the auto refresh timer is expiring and refreshing the page. That clears the selected files markers.

MartinMueller2003, 2023-08-22:
Fix in PR#652

New dist at
https://drive.google.com/drive/folders/1bGrkXSMoGjWtX5iWspPFRogM2G6c1VXu?usp=sharing

closed 2023-08-22 by MartinMueller2003

cybercop23, 2023-08-22:
Hi Martin. Used this dist and flashed the DigUno/AE+ (see below) and still can't remove the file. 
[image]
Please LMK what else I can provide.
14:25:35: [FileMgr] Courtesy Of The Red White And Blue.fseq - 7982
14:25:35: [FileMgr] Dance Monkey.fseq - 4308
14:25:35: [FileMgr] Danger Zone.fseq - 8765
14:25:35: [FileMgr] Fancy Like.fseq - 3357
14:25:35: [FileMgr] Firework.fseq - 9245
14:25:35: [FileMgr] God Bless the USA xTreme.fseq - 6832
14:25:35: [FileMgr] I Love My Country.fseq - 3833
14:25:35: [FileMgr] USA-fill.fseq - 668
14:25:35: [FileMgr] BatsTest.fseq - 42904
14:25:35: [FileMgr] 2022_The Devil Went Down to Georgia.fseq - 57285496   <== was tryin  g to remove this one
14:25:35: [FileMgr] test_aeplus_upload.fseq - 90072

MartinMueller2003, 2023-08-22:
2nd fix added to PR#652. Had forgotten to implement file delete  when I removed WS. Had a nice big todo on it that I missed.

MartinMueller2003, 2023-08-22:
dist.zip has been updated

cybercop23, 2023-08-22:
All good. File deletion, one at a time, or multiple. works.
Will test the other change after the nightly xlights.. pr already merged.
There are some FPP/proxy changes that may affect IDing ESPx but we'll see
later.

Thanks again for all your support and help.