forkineye/ESPixelStick #653: Create optional Web installer similar to WLED

patdelaney, 2023-08-24:
**Is your feature request related to a problem? Please describe.**
The current ESPixelStick installer does not work on M1 or M2 Macs. (so I have been told)

**Describe the solution you'd like**
Create a 2nd install option for now that is a Web installer similar to what WLED is doing.
https://install.wled.me/

**Additional context**
Keep the Java one for now until we get all the bugs out of the web installer.

patdelaney, 2023-08-24:
This may be similar to #513

marcobrianza, 2024-01-05:
I support this request because moved to a MAC M1 and cannon make the java installer work

forkineye, 2024-01-05:
> I support this request because moved to a MAC M1 and cannon make the java installer work

Use the latest beta and run it from the command line. You'll need to follow the MacOS pop-ups to allow the binaries to run - https://github.com/forkineye/ESPixelStick/releases

marcobrianza, 2024-01-05:
Yes it worked thank you. I executed in terminal:
 java -jar ESPSFlashTool.jar

MartinMueller2003, 2024-03-27:
WLED is using a package from ESP Home Automation. Looking into what it will take for PixelStick to use it as well. We will need to host the server somewhere.

https://github.com/esphome/esp-web-tools

forkineye, 2024-03-28:
I've considered doing something like this, however we do things a little differently when flashing. Instead of just baking an image, we generate the filesystem with the user's credentials and flash it as well.  Although possible, I'm not comfortable asking for people's wifi credentials to generate an image on the backend of a web service.  In order for this to work and keep such things all client side, we need to implement a CLI and the Improv stuff to configure the stick post-flash.  I can host the service on aws.

MartinMueller2003, 2024-03-29:
Combining the images after the build to create a monolithic Code + FS image is not that difficult. We can still create the config file in the same way the tool works by using a web page that has a button to the web-tools upload function. Add in an interface that allows us to upload the config.json file via the serial interface and we are good to go.

Process Steps:

- User accesses our web site
- fill in the same data you use today.
- save the config file to the local disk
- upload the image to the device.
- send the config file to the device.

Those are my thoughts at the moment. It is likely this will change as I dig deeper into the details.

MartinMueller2003, 2024-04-02:
Command to combine images on a windows system. Creates an image that is the size of the flash file. Final image is uploaded starting at address 0x0

.\dist\bin\esptool\esptool.exe --chip esp32 merge_bin  -o .\firmware\esp32\esp32_ttgo_t8-merged.bin --flash_mode dio --flash_freq 80m --flash_size 4MB 0x0000 .\firmware\esp32\esp32_ttgo_t8-bootloader.bin 0x8000 .\firmware\esp32\esp32_ttgo_t8-partitions.bin 0xe000 .\firmware\esp32\boot_app0.bin 0x10000 .\firmware\esp32\esp32_ttgo_t8-app.bin 0x003b0000 .\firmware\esp32\esp32_ttgo_t8-littlefs.bin

Wrote 0x400000 bytes to file .\firmware\esp32\esp32_ttgo_t8-merged.bin, ready to flash to offset 0x0

MartinMueller2003, 2024-04-15:
Still making progress. I have the image with custom config built and a manifest created. Just starting integration with the framework tool. Going to initially use the version hosted by the framework. Once working I will move it to a self hosted model. You can see latest code at: https://github.com/MartinMueller2003/EspsWebFlashTool

MartinMueller2003, 2024-04-15:
Also, the server is developed as a nodejs server. Going to need to modify that to work with the forkineye server.

MartinMueller2003, 2024-04-16:
I just flashed my first ESP V4 using a web based flash tool. :)

MartinMueller2003, 2024-05-07:
Video here on how to use the web flasher.
https://www.youtube.com/watch?v=olu29-oUXTo
And the url for the tool: https://espixelstickwebflasher.from-ct.com:5000

marcobrianza, 2024-05-08:
I just flashed a D1 mini for breakfast and start a great day :-)
very good job thank you!

100ideas, 2024-05-30:
Had some struggles with odd non-ascii serial output in the java app on osx intel macbook while trying to flash an espixelstickv3. I came over here and was delighted by this issue - thanks! I gave it a shot and it seems to have worked. Thanks for working on this!
[image]