forkineye/ESPixelStick #654: ESPixelStick Flash Tool on Mac

cybercop23, 2023-08-25:
**Is your feature request related to a problem? Please describe.**
Currently the Flash Tool doesn't work on the M1/M2 Macs because the librxtxSerial.jnilib is not supported being older version.

**Describe the solution you'd like**
Possibly include the updated lib in the Jar file or better yet use the wled web install process to flash the devices.

**Additional context**
For reference: https://install.wled.me/

forkineye, 2023-09-19:
Fixed as of e37fd855746747f5f5ea62e6b2c97b2e645673c5

closed 2023-09-19 by forkineye

forkineye, 2023-09-19:
> Thank you @forkineye for this fix. Maybe I'm doing something wrong, which is always possible, but AFAIK on the M1/2 you can't install python v2. I created an alias and I can run it from the terminal with:
> 
> `python bin/upload.py --chip esp32 --baud 460800 --port /dev/cu.usbserial-110 --before default_reset --after hard_reset write_flash -z 0x1000 firmware/esp32/esp32_quinled_uno_ae_plus-bootloader.bin 0x8000 firmware/esp32/esp32_quinled_uno_ae_plus-partitions.bin 0xe000 firmware/esp32/boot_app0.bin 0x10000 firmware/esp32/esp32_quinled_uno_ae_plus-app.bin 0x3B0000 firmware/filesystem.bin`
> 
> However the Flashing tool, still tries to use python and doesn't have access to the .profile to use the alias.. so it fails... `-= Erasing ESP Flash =- Command: python bin/upload.py --chip esp32 --baud 460800 --port /dev/cu.usbserial-110 erase_flash exec - Start java.io.IOException: Cannot run program "python": error=2, No such file or directory`
> 
> Anyway to detect if a Mac M1/M2 and issue python3 or maybe have a button (default) to python and have the option to choose python3?

This is related to https://github.com/forkineye/ESPixelStick/issues/562.  I need to add some logic to detect supported python binaries as java doesn't access the alias as you've noticed. Can you use a symlink for now?