forkineye/ESPixelStick #655: Sequence uploaded via fpp connect gets odd timestamp when viewed in File Management

patdelaney, 2023-08-26:
**ESPixelStick Firmware Version**
FW Version4.0-ci5943922254
Build DateAug 22 2023 - 20:55:51

**xLights Version**
2023.13

**Hardware Version**
Shelby's V3

**Binary release or compiled yourself?**
CI Build

**Operating System (and version)**
esp8266

**Web Browser (and version)**
Chrome

**Describe the bug**
I created a sequence in FPP and used FPP connect to send it to the ESPixelstick and the time stamp for the newly created file was 12/31/2097, 6:11:12 PM

 See Attached screenshots
FPP Connect
[image]

ESPixelstick
[image]

cybercop23, 2023-08-26:
WOW.. that's wo weird. I get 1979.... Other user does seem to get the right time on that same hardware.
I'm using a D1 Mini32.
[image]

MartinMueller2003, 2023-09-17:
Keep in mind, the ESP has no idea what time it is unless you tell it the current time. Opening the web UI causes the current browser time to be sent to the ESP and your timestamps will be ok. If you boot (default time is 1/1/1970) and then have FPP Connect upload files you will get some timestamp in that region.

closed 2023-10-05 by MartinMueller2003