forkineye/ESPixelStick #656: Device Setup - Changes not saved on QUINLED QUAD flavor

patdelaney, 2023-08-27:
**ESPixelStick Firmware Version**
FW Version4.0-ci5943922254
Build DateAug 22 2023 - 20:53:34

**Hardware Version**
esp32_quinled_quad

**Binary release or compiled yourself?**
CI Build

**Operating System (and version)**

**Web Browser (and version)**
Chrome

**Describe the bug**
I'm testing the QUINLEF QUAD with the the normal hat (no SD not ethernet, no extra ports)
When it boots up I see the following errors in the log and I am not sure if they can be ignored

21:15:24: [FileMgr] ***Configuration File: '/output_config.json' Deserialzation Error. Error code = IncompleteInput***
21:15:25: [FileMgr] Configuration File: '/output_config.json' saved 6787 bytes.
21:15:30: [FileMgr] Configuration File: '/admininfo.json' saved 176 bytes.
21:15:30: [FileMgr] Configuration File: '/config.json' saved 370 bytes.
21:15:30: [FileMgr] Configuration File: '/output_config.json' loaded.
21:15:30: [WiFiDrv] WiFi Entering State: Connecting Using Config Credentials
21:15:31: [WiFiDrv] Using DHCP
21:15:31: [WiFiDrv] Connecting to 'paddycakes' as quinled-2
21:15:31: [WiFiDrv] WiFi Entering State: Connected To AP
21:15:32: [WiFiDrv] Connected with IP: 192.168.1.145
21:15:32: [ E1.31] Listening for 0 channels from Universe 1 to 0 on port 5568
21:15:32: [WebMgr] Web server listening on port 80
21:15:32: [  FPPD] Listening on port 32320
21:15:32: [ 16987][E][vfs_api.cpp:104] open(): /littlefs/www/index.html does not exist, no permits for creation

[error.log](https://github.com/forkineye/ESPixelStick/files/12447100/error.log)

If I configure E1.31 as an input and then some outputs, it does not save the config. IF I select DDP as input it does save the config.
[image]

It's very random

patdelaney, 2023-08-31:
I think that the issue maybe time timing related. I think I am trying to setup the device before it's completely stable. I'll try to capture a video of what is happening.

MartinMueller2003, 2023-09-17:
By default, that initial boot never has any config files. The config files are auto generated on boot if a version is not found on the file system. That means that if the config files ever get corrupted, then default operation will be maintained. So if those errors were the first boot after flashing, then they are expected. Below is a snippet of my first boot after clearing the config. You get the same thing if you click the factory reset on the admin screen.

[  4357][E][vfs_api.cpp:104] open(): /littlefs/config.json does not exist, no permits for creation
[  4362][E][vfs_api.cpp:104] open(): /littlefs/input_config.json does not exist, no permits for creation     
[FileMgr] Configuration File: '/input_config.json' saved 1963 bytes.
[  5136][E][vfs_api.cpp:104] open(): /littlefs/output_config.json does not exist, no permits for creation
[FileMgr] Configuration File: '/output_config.json' saved 6787 bytes.
[FileMgr] Configuration File: '/admininfo.json' saved 167 bytes.
[FileMgr] Configuration File: '/input_config.json' loaded.
[FileMgr] Configuration File: '/config.json' saved 358 bytes.
[FileMgr] Configuration File: '/output_config.json' loaded.
[WiFiDrv] WiFi Entering State: Connecting Using Default Credentials
[WiFiDrv] Using DHCP

MartinMueller2003, 2023-09-17:
The more interesting issue is:

MartinMueller2003, 2023-09-17:
I just did a bunch of config changes and am not seeing issues.

MartinMueller2003, 2023-10-29:
Please test with https://drive.google.com/drive/folders/1bGrkXSMoGjWtX5iWspPFRogM2G6c1VXu?usp=sharing. While no config specific changes have been made, I have been saving the config many times and not seeing anything. The annoying thing I see is that the save is now so fast that the :saving" message on the UI does not show.