forkineye/ESPixelStick #657: espixelstick stops functioning after about 5min

stijn-dhondt, 2023-09-03:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
4.0-ci5943922254

**Hardware Version**
quinled quad ETH
but have same issues with Uno and Octa

**Binary release or compiled yourself?**
release

**Operating System (and version)**
Windows

**Web Browser (and version)**
Version 114.0.5735.199 (Official build) (64-bits)

**Access Point**
Ubiquity unifi UAP-AC-Pro

**Describe the bug**
ESP 32 boots comes as available controller in FPP player but after about 5min controller goes offline and does not respond to anything anymore. Until i power cycly the device it stay's offline. 
As from that point the same issue just repaets.

even if i send ping to the device so it stays awake it just stops functioning. 
Also no response on the console just no more comms..

stijn-dhondt, 2023-09-06:
Xlights 2023.13 on mac 
FPP 7.1 git 5f5cfcc
I already tested the firmware on following Quin boards:
- Dig Uno wifi only
- Dig Quad eth and wifi
- Dig Octa eth and wifi

all dropping all of the communication within less than 5min...
Don't think Xlights and FPP are the issue here it's really the ESP that completely stops responding..
Even the console does not output anything...

rledonne, 2023-09-06:
I believe I am having a similar issue that may be the same thing.  I added comments as a discussion.
https://github.com/forkineye/ESPixelStick/discussions/659

It will stay connected until you open Tools | Test in xLights and start to send it data.  Then it goes offline and doesn't reconnect until power cycled.

rledonne, 2023-09-06:
Adding an update.  
I noticed that the network connection will drop randomly and not reconnect after a few minutes, regardless of whether I send data or not.  Sending data to it just makes it happen faster.

If I add a fseq to the sd card and play it from the firmware, it will continue playing and looping after the network connection drops.

I am available to troubleshoot, I just don't know how to get any useful information since it a wifi device, and no logging, or maybe I am missing it.

stijn-dhondt, 2023-09-06:
> Adding an update. I noticed that the network connection will drop randomly and not reconnect after a few minutes, regardless of whether I send data or not. Sending data to it just makes it happen [...]

Turned off DDP and did not solve the issue on all 3 controllers and no change so DDP is not the source of the issue here.
Seem that we really are looking at a similar issue with your update now. Great to not be alone here ;-)

stijn-dhondt, 2023-09-12:
@patdelaney are you able to reproduce the issue?

stijn-dhondt, 2023-09-18:
found version 4.0-ci3677767017 on an older system and this one seems to work normal... but that's one of 12 dec 2022.

MartinMueller2003, 2023-09-18:
Please try this version: https://github.com/MartinMueller2003/ESPixelStick/blob/main/dist.zip

Please leave  the ESP connected to the flashtool. Do whatever it is you normally do to cause it to disconnect. Save the logs and attach them to this issue.

stijn-dhondt, 2023-09-18:
[Log.txt](https://github.com/forkineye/ESPixelStick/files/12651508/Log.txt)

@MartinMueller2003, Logs attached... I started the esp arround 18:56:30 and it went offline from wifi today at 19:03:50 but no logs appeared at that time...

MartinMueller2003, 2023-09-18:
Interesting that the wired interface came up but never got an IP address. That may be causing an internal routing conflict.  Can you post your network config screen?

stijn-dhondt, 2023-09-18:
[image]

This is the last config i tested, before i had DHCP turned on with the same issue, and i also tested with the same ip on the ethernet interface which should be an issue anyway but in all 3 cases there was nothing connected to the ethernet interface...

MartinMueller2003, 2023-09-18:
I never tested With nothing plugged into the Ethernet port and the IP set to static.

rledonne, 2023-09-18:
I will add mine as well....
[AEplus Log.txt](https://github.com/forkineye/ESPixelStick/files/12652547/AEplus.Log.txt)

This was after a fresh flash using the above zip.
It flashed fine, connected to wifi, and watching it in the router, was connected for about 30 seconds before it dropped.  During that 30 seconds, I clicked the output to lights in xlights, and started a sequence.

stijn-dhondt, 2023-09-18:
@MartinMueller2003 just had 1 time the exact same issue in the logging as @rledonne! other times the console just stays silent...

rledonne, 2023-09-20:
This one was flashed and nothing was done after it was flashed yesterday afternoon.  It just sat connected to the computer.  It looks like it dropped overnight and never tried to reconnect.

[AEplus Log2.txt](https://github.com/forkineye/ESPixelStick/files/12672675/AEplus.Log2.txt)

MartinMueller2003, 2023-09-20:
Now that is MUCH more interesting. These three lines are odd:
This line indicates the web page asked for the files needed to display and started working
16:19:31: [ 32710][E][vfs_api.cpp:104] open(): /littlefs/www/esps.css does not exist, no permits for creation

This line indicates there was a problem working with the AP. This one is VERY odd because the ESP should have stopped transferring data right away.
16:19:50: [WiFiDrv] WiFi Lost the connection to the AP

And this is when the ESP finally gave up interacting with the AP.
03:03:17: [WiFiDrv] WiFi Entering State: Connection Failed

Now to try and replicate what is going on. FYI: I ran mine overnight last night with no issues.

rledonne, 2023-09-20:
Sorry, I just remembered this.
I think the line 16:19:31: [ 32710][E][vfs_api.cpp:104] open(): /littlefs/www/esps.css does not exist, no permits for creation happened when I did go to the Network Settings page to make sure it was connected and since you had asked for a screenshot earlier, but there was nothing really there, just the DHCP box and AP Reboot checked.

It did sit  with "16:19:50:                                     "  in the serial output while waiting.  My guess is when it dropped around 3am, it finished that line with  [WiFiDrv] WiFi Lost the connection to the AP

rledonne, 2023-09-20:
Please let me know if there is any other way that I can help.

stijn-dhondt, 2023-09-24:
just a quick update!
I started using the latest version and had similar issues.
I removed the FPP 7.1 from the network for now just to eliminate possible sources and all 3 controllers (Dig Uno, Quad and octa remaining online for now. so it sees that at first site FPP is causing the issue. 

Will downgrade FPP to a stable version later today. @rledonne which fpp version are you running?

MartinMueller2003, 2023-09-24:
Excellent. That is something I can work with.

stijn-dhondt, 2023-09-24:
> Excellent. That is something I can work with.

@MartinMueller2003 6.3.2 seems to be stable  except for when the controllers are connected trough ethernet they show up in FPP with ip 0.0.0.0

[image]

MartinMueller2003, 2023-09-24:
Added FPP 7.1 to my network (I stopped at 5.3 because I don't agree with the new hat license) and I see the heap dropping regularly. After about 5 minutes the heap drops to a level where no network traffic gets through the system. Only happens when FPP 7 is sending updates. Time to start debugging :)

MartinMueller2003, 2023-09-25:
Please try the dist.zip found here: https://drive.google.com/drive/folders/1bGrkXSMoGjWtX5iWspPFRogM2G6c1VXu?usp=sharing

stijn-dhondt, 2023-09-26:
@MartinMueller2003 just tested a dig quad trough wifi and ethernet and working fine now!

Do you want me to test a dig uno and Octa also?

MartinMueller2003, 2023-09-26:
That would be good but the code is common so if it works on one it should work on the others.
Martin

MartinMueller2003, 2023-09-26:
I fixed a memory leak in the file close process. It was possible to miss closing a file.

stijn-dhondt, 2023-09-26:
Still have some strange issues but give me some time to come back trying to pinpoint it first...

stijn-dhondt, 2023-09-27:
@MartinMueller2003 Looking good for now! Controller has been online the whole night. Can you let us know when the update will be available in CI build?

MartinMueller2003, 2023-09-27:
Fixed in: [Fixed memory leak](https://github.com/forkineye/ESPixelStick/pull/668#top)
#668

closed 2023-09-27 by MartinMueller2003