forkineye/ESPixelStick #658: playing a file .seq file does not consider universe / channel

marcobrianza, 2023-09-06:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
4.0-ci5890809791

**Hardware Version**
Dig Octa

**Binary release or compiled yourself?**
CI release 

**Describe the bug**

I have a setup with 2 dig octa,
they are using one universe each (1 and 2)
all 8 outputs are used on each controller. 
It works  beautifully in Arnet streaming via xLights.

I have uploaded the same .seq file exported by xlights to both controllers.
I trigger the files by a MQTT message.

The first controller on universe 1 plays the correct pixels data.
The second controller does not consider to play the shifted data from universe 2 and still plays the universe 1 (at least it what look like to me).

Am I doing things wrong or is this a bug?

MartinMueller2003, 2023-09-06:
The .fseq file for each ESPixelStick needs to be unique and MUST only have the data for the channels on the stick.

marcobrianza, 2023-09-06:
Thank you for the clarification, do you know a way to export separate files from xLights relative to each controller?

MartinMueller2003, 2023-09-06:
Sorry, but no. I am a vixen user. It is my understanding that a feature called "FPP Connect" does this but I do not know for sure.

MartinMueller2003, 2023-09-06:
Keep in mind that triggering the fseq file from mqtt results in two individual ESPs running the file. They may not be exactly in sync. The mqtt trigger may be delayed and temperature affects the ESP clocks so even if they start at the same time, they could become misaligned. This is resolved by triggering an FPP master and having it control the FPP remotes (ESPs). Then they will be in sync.

marcobrianza, 2023-09-06:
Ok thank you I will try that. 
At the moment I don't see the espixelstick controllers, maybe because I have configured Artnet+MQTT
do I need to configure FPP-remote instead of MQTT?

last time I enabled MQTT all the ports config was reset...

MartinMueller2003, 2023-09-06:
If you are using V4, The output port config is seperate from the input config. However, xLigths may clear the xLights configuration which I cannot do anything about.

marcobrianza, 2023-09-06:
Im using Version4.0-ci5890809791
Build Date Aug 17 2023 - 12:23:57

I have seen that there is a new releases that fixes some issues with config files, (it might fix my issue base because switching from FPP remote to MQTT resets the outputs that are in the same page)  

I cannot find the .efu files required by the web interface to update the firmware, I just see the .bin files in the archive.

In the meantime I had enabled FPP remote on a test controller and xlights had uploaded a file that if half size of the original one, so I think that it is taking into account the specific controller data (I can verify tomorrow and will update)